PrecisionGazeMouse/PrecisionGazeMouse #1: Move mouse only when triggered

nash8350, 2017-07-23:
Suggestion from Jason's Random Videos on YouTube:
"Hi Henry. I love your idea for using eye-tracking for larger mouse movements and head tracking for smaller, more precise mouse control. My only concern is that it's "always on". I don't like the mouse following me around as I read, for example. That is why I like that the Tobii 4C only moves the mouse to where I'm looking when I click on a trigger key. If you could incorporate something similar, I think it would be a huge upgrade to your product. For example, you could set it up so that the cursor only moves with eye tracking when you click a trigger, but the head tracking is always on. Thoughts?﻿"

Yes this should be possible. We could add a hotkey that moves the mouse. Leaving this open to see if other people like this suggestion as well.

nash8350, 2017-08-10:
I really liked how Clippy handled this and so I added support for moving the mouse on a key press

closed 2017-08-10 by nash8350

trishume, 2017-08-11:
The way I handled this in https://github.com/trishume/PolyMouse is that eye tracking is only used to accelerate head motions. So if you don't move your head the mouse doesn't move.

If you start a significant head movement, and you are looking far from the cursor, then it begins an animated warp.

nash8350, 2017-08-15:
Thanks @trishume I opened #2 to track this