PrecisionGazeMouse/PrecisionGazeMouse #3: application not opening

nosaj85, 2017-08-16:
I downloaded the latest release (v1.02), extracted the files in the folder, and ran the application file (as both a user and administrator). However, nothing happened when I tried to run it. I have the Tobii eye tracker 4c, but not the TrackIR.

nash8350, 2017-08-19:
Thanks for trying it @nosaj85! I think it should work with just the Tobii eye tracker, since the TrackIR is not required. What version of Windows are you running on? I am using Windows 10.

If it's not too hard, it would also be good to know what version of .NET you're using. You can check by opening the file Explorer and navigating to %windir%\Microsoft.NET\Framework\. For me the directory is C:\Windows\Microsoft.NET\Framework. Then I look for the folder with the highest version. For me it's v4.0.30319.

nosaj85, 2017-08-19:
hey @nash8350. I'm also using Windows 10, and also .NET version v4.0.30319. 

I downloaded v1.02 and unzipped it. The unzipped folder has 5 dll files and 1 application file. When I run the application file, my computer thinks for a moment, but nothing happens (nothing opens; nothing shows in my taskbar; in task manager, no new programs are opened and I don't see any background processes that seem related to it).

nash8350, 2017-08-23:
I just tried downloading the zip file from GitHub and running it myself, and I noticed that a pop up from Windows defender smart screen opened and asked me if I wanted to run a file from an unknown publisher. is it possible that your smart screen is configured to block unknown publishers? I believe you can check your smart screen settings in the control panel.

I really should make a proper installer for this instead of just giving a zip file. I just haven't had a chance yet, but I'm creating a separate issue to track creating an installer #4.

nosaj85, 2017-08-23:
I have the pop-up and allowed it to run. After that, my computer thought for a moment (cursor became a turning wheel), but nothing happened after that.

nash8350, 2017-08-23:
Okay so that's not it! Another thing to check is the Windows event log. In the start menu open the Event Viewer program. Click on Windows logs > Application > Find. Enter "Gaze" and maybe you will see one or more results? If you see anything, could you paste it?

nosaj85, 2017-08-23:
Here are 2 errors in the log... 

Source: .Net
Event ID: 1025

Application: PrecisionGazeMouse.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at TrackIRUnity.TrackIRClient.GetDLLLocation(System.String ByRef)
   at TrackIRUnity.TrackIRClient.TrackIR_Enhanced_Init()
   at PrecisionGazeMouse.PrecisionPointers.TrackIRPrecisionPointer..ctor(PrecisionGazeMouse.PrecisionPointers.PrecisionPointerMode, Double)
   at PrecisionGazeMouse.MouseController.setMode(Mode)
   at PrecisionGazeMouse.PrecisionGazeMouseForm..ctor()
   at PrecisionGazeMouse.Program.Main()

Source: Application Error
Event ID: 1000

Faulting application name: PrecisionGazeMouse.exe, version: 1.0.0.0, time stamp: 0x598c8be7
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x0088f516
Faulting process id: 0x1e14
Faulting application start time: 0x01d318d0be4e0a59
Faulting application path: C:\Users\XXX\Downloads\Precision.Gaze.Mouse.v1.02\PrecisionGazeMouse.exe
Faulting module path: unknown
Report Id: a9f0a504-9657-4f98-9d52-57d640ede170
Faulting package full name:

nash8350, 2017-08-25:
Thanks that's very helpful! It's looking for a registry value that is not present when the track IR is not installed. I have fixed the bug in the latest release v1.03. Can you try that version?

nosaj85, 2017-08-26:
Good news! it opens now. I'll provide some additional feedback in new issues.

closed 2017-08-26 by nosaj85