PrecisionGazeMouse/PrecisionGazeMouse #5: Incorporate head rotation when key-press is on

nosaj85, 2017-08-26:
I would like to be able to place the mouse using my gaze, with a trigger key, and then be able to make more precise movements around that placement using head rotation. Currently, when placement is made based on a key press (for gaze), head tracker has no impact on movement of the mouse. This could be fixed by having the option of toggling the movement for gaze position to be based on continuous/key-press, and separately toggling movement for head rotation to be continuous/key-press/off.

nash8350, 2017-08-28:
It allows me to rotate my head. Are you holding the trigger key until you have finished placing the mouse?

nosaj85, 2017-08-29:
The cursor does not continue to move if I hold down the trigger key (based on eye movement or head rotation). Though the idea of being able to continuously move the mouse by holding down the trigger key is something that would be really nice (even in eye-tracking only mode).

nash8350, 2017-08-29:
It sounds like maybe there is a bug because that is how it is supposed to work. Let's see if we can get this figured out! I assume you're in EyeX only mode, with On Key Press movement? What key are you using? Do the gaze position and head rotation fields display two numbers and do they update continuously? Do they update under Continuous movement mode? They don't when using On Key Press after you press and hold the trigger key? What does the status field say when you do this?

nosaj85, 2017-08-29:
Interesting... Click and hold did not work with F3 as the trigger, but it is working with any other key. If I set the trigger to anything else, I can hold it down and move the mouse with both gaze and head rotation. I'm guessing F3 was an issue because it's also what I have the Tobii software using as a trigger

closed 2017-08-29 by nosaj85