PrecisionGazeMouse/PrecisionGazeMouse #6: Head Rotation is too sensitive

nosaj85, 2017-08-26:
I would like to be able to place the mouse using my gaze, with a trigger key, and then be able to make more precise movements around that placement using head rotation. Currently, head rotation sensitivity is very high. That is, fairly small head movements lead to large mouse movements, making it difficult to make accurate small-distance adjustments to the cursor after using gaze to place it. This could be remedied by either making head rotation less sensitive, or by including a drop-down menu allowing the user to select their desired sensitivity.

trishume, 2017-08-26:
Something I implemented in http://github.com/trishume/PolyMouse that might be helpful is an acceleration function (like mouse cursor acceleration) for the head tracking. So if you make slow head movements it moves the cursor slowly, but faster movements still allow you to move larger distances.

nash8350, 2017-08-28:
I like the suggestions of having a sensitivity setting on the UI and potentially having an acceleration function. These will require code changes. My guess is the sensitivity setting on the UI will be easier to implement as a first step. In the meantime you can adjust the sensitivity in the code here https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/PrecisionPointers/EyeXPrecisionPointer.cs#L69

nash8350, 2017-10-11:
@nosaj85 I added the sensitivity adjustment in v1.04 check it out! Let me know if you run into any issues. Closing for now. @trishume I'll move your suggestion over to a separate issue.

closed 2017-10-11 by nash8350