PrecisionGazeMouse/PrecisionGazeMouse #7: Turn off automatic clicking when set to "on key press"

nosaj85, 2017-08-26:
I would like to be able to place the mouse using my gaze, with a trigger key, and then be able to make more precise movements around that placement using head rotation, and then click if I choose to. Currently, when I have movement based on key press, the cursor seems to be automatically left click whenever I place the cursor (before I even have a chance to move the cursor with head rotation). The easiest solution is to turn this off (you could use an AHK package to also designate a key to left click). A more ideal solution would be for you to have a set of options built in that will allow the user to designate certain keys to acts as left click, right click, hold down left click, release left click (for click and drag).

nash8350, 2017-08-28:
Did you try holding the trigger key down until you were done positioning your mouse? It only clicks on key up. This should allow you to rotate your head to precisely position the mouse before clicking. I set my trigger key to F3 and then have AHK execute that as a left click or left click and hold. See the documentation for this mode at http://precisiongazemouse.com. 

I don't usually have a need to move my mouse without clicking. If you want to continuously move the mouse without clicking there is the continuous mode. Can you describe a scenario where you would want to move on a trigger key but not click the mouse?

nosaj85, 2017-08-29:
The main reason I currently don't want the automatic click is that I'm not yet able to place the cursor perfectly, and the automatic clicking can lead to some very frustrating misclicks. Automatic clicking can also make it more challenging to click and drag, or when I want to right click on something without left clicking first (e.g., if I have a group of items selected and I want to take an action on all of them that requires a right click, left clicking first may un-select the group).

Tobii's software defaults to having the trigger key also click (which I'm guessing most people prefer), but there's a toggle to turn it off.

nash8350, 2017-08-29:
Okay I guess that makes sense! It does have support for click and drag you down trigger key down-up-down and hold to drag. You make a good point for right clicking though. Usually when I want to right-click I hold down the trigger key to keep the mouse moving, say "right click" to my dictation software, and then mouse up when I'm ready to left click. This is a litle complicated though.

What do you think about adding a mode where we can choose between click behaviors like normal click, trigger key click, or none?

nash8350, 2017-10-11:
@nosaj85 I added the ability to specify a separate hotkey for clicking. You can also leave the box blank to turn it off. Check it out in v1.04! Let me know if you run into any issues. Closing this for now.

closed 2017-10-11 by nash8350