PrecisionGazeMouse/PrecisionGazeMouse #8: Interference between TrackIR & Tobii

shervinemami, 2017-09-25:
Hi,

I've got a TrackIR 5 and a Tobii Eye Tracker 4C, I installed the drivers for both, and verified that both of them work correctly with their official software. But when I run the prebuilt PrecisionGazeMouse v1.0.3 release on Windows 10, it detects my Tobii fine but it doesn't detect my TrackIR, it says there's no connection to TrackIR. If I run the official NaturalPoint TrackIR v5 software it shows correct head tracking in that software, but not in PrecisionGazeMouse. I also tried using your older prebuilt PrecisionGazeMouse v1.0.1 release but it has the same problem.

shervinemami, 2017-09-27:
I'm not sure if it's related to my problem, but on the PrecisionGazeMouse.com website it says that in the TrackIR software to choose “GazeMousePlus” in the titles tab. But GazeMousePlus isn't listed in the titles tab. I'm using TrackIR v5.4.2 and Precision Gaze Mouse v1.0.3.

nash8350, 2017-09-29:
So normally I run the TrackIR program first and check to make sure it's working in the official software. Then I open up PrecisionGazeMouse at the same time. At this point, the TrackIR screen gets minimized to the tray but continues running in the background. Can you verify yours continues running in the background?

shervinemami, 2017-09-30:
After clicking on "Game Updates", I can see GazePlusMouse, and using it, PrecisionGazeMouse now works, thanks!

Also I'd like to ask you for some customization, can you please email me at "shervin.emami@gmail.com" so we can discuss a bit of paid support?

closed 2017-09-30 by shervinemami