PrecisionGazeMouse/PrecisionGazeMouse #10: GUI doesn't show up

Galothus, 2017-11-24:
After starting the executable the GUI won't show.
Tried it on Windows 10 version 1511 ( didn't check .NET version) and 1703 with .NET Framework 4.7

Right now I'm waiting for Windows to finish updating to 1709. Maybe that will help?

nash8350, 2017-11-28:
Thanks for trying it out! I tried extracting then running from the zip file for version 1.05 and it worked for me. I'm also on 1709 with .NET 4.7.02556. Another user reported that it works like a charm for him. Are you using v1.05? Do you see any error message? You might be able to check for messages in the Event Viewer under Windows Logs > Application > Find "Precision"

Galothus, 2017-12-04:
I get these two errors.

1[image]
[image]

nash8350, 2017-12-08:
Looks like it can't load a DLL dependency. Can you confirm that Tobii.Interaction.Net.dll, Tobii.Interaction.Model.dll, and Tobii.EyeX.Client.dll are extracted and in the same directory as PrecisionGazeMouse.exe? They won't load if you run from the zip file. Also, is the Tobii EyeX program installed and running? Lastly, do you have a 64-bit computer?

nash8350, 2018-01-11:
I haven't been able to reproduce this problem and it's been over a month since I heard from the reporter. If anyone else runs into this issue, please let me know and we can reopen it.

closed 2018-01-11 by nash8350