PrecisionGazeMouse/PrecisionGazeMouse #11: Trojan in version v1.06.zip

schwabdidier, 2018-02-03:
First of all, thank you very much for your work. I think that your tool will help lots of people (especially children) with strong disabilities to access communication.

While the 1.05 works perfectly well, my windows 10 annonces a Trojan with the version 1.06.

nash8350, 2018-02-06:
Hi I'm glad you found a use for it! What program reported the trojan and what version is it? I'm also running Windows 10. I downloaded the zip and scanned it using Windows Defender and it said "No threats found". I also checked using Kaspersky and didn't find anything either https://virusdesk.kaspersky.com. 

We can check if the version of your file is the same as mine. I opened powershell and ran this:

```
> Get-FileHash .\Precision.Gaze.Mouse.v1.06.zip -Algorith
m MD5

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
MD5             8313EA8B93B9334A72780C2497D91DE6
```

If the hash matches on your version then the file is the same. Otherwise it's been altered.

schwabdidier, 2018-02-06:
Unfortunately I will be able to check that only on Friday. It was a Windows 10 which was just installed which warn me.

nash8350, 2018-02-08:
Ok since its a fresh install of Windows make sure to update Windows and the virus checker tool. Then see if it still complains.

schwabdidier, 2018-02-10:
No, you are right it is ok, I can't reproduce the problem and now I have last released version. Sorry, and again thank you for your work !

closed 2018-02-10 by schwabdidier