PrecisionGazeMouse/PrecisionGazeMouse #12: Great app - and a couple of ideas

Nbech, 2018-02-06:
First I want to say that I am exited to have found this app.
I have a sister who is suffering from Sclerosis, and therefor unable to use her hands and having increasing speech difficulties. This is just what I have been looking for to help her, and at an affordable price. 
After a quick test, I have a couple of thoughts/ideas for adjustments of the app. I am a developer myself, so I would love to contribute to the development. I am new to github though, so I need to explore this platform a bit before doing something useful :-)
Anyway, here is the thoughts and ideas:  
- Is it possible to adjust the size of the gaze tracker circle?
- It would be nice with an option to let the mouse click automatically, if the focus is kept in the same spot for a given timeperiod.  
- If I choose EyeX Only, the head rotation is not detected. Head rotation is only detected if I choose to combine it with eViacam. (head rotation is detected by the tobii software, but not by PGM).
- I have downloaded a program called Communicator 5 from tobii, but when this program is active, the cursor is not controlled by PGM, but can only be controlled by the regular mouse. (I'm not sure if this is caused by the fact that I am only running an evaluation version of the communicator program, but it works fine with eViacam and head movement control only, if I disable tobii C4 and PGM).

nash8350, 2018-02-08:
1. It is possible to adjust the size of the gaze tracker circle in the code here https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/WarpPointers/EyeXWarpPointer.cs#L21. 
2. I believe this is called Dwell clicking. eViacam supports it, but no one has built it into PGM yet.
3. You have the EyeX 4C version right? It is only supported in that version. Rotation works for me in EyeX only mode but the control is not very smooth. You can try adjusting the sensitivity higher using the control bar. Let me know if it still doesn't work.
4. It`s possible that communicator 5 is running as an administrator. PGM runs in user mode and is not able to control the administrator windows. To get around this you can try right clicking the program and then clicking run as administrator.

schwabdidier, 2018-02-10:
@Nbech : I use PrecisionGazeMouse in a similar way for my daughter. I don't have communicator but I will try the evaluation version soon. There is a free software called GazeSpeaker (http://www.gazespeaker.org), I think it would be easily usable for your sister. It is a fantastic work but the guy who did that don't answer to my mails, I really would like to see the code on GitHub to improve it.

Nbech, 2018-02-11:
Your guess was right: PrecisionGazeMouse works with Communicator, when PGM is started in administration mode.
I have now created an input field in the app, so it is possible to resize the gaze tracker circle.
I have also added dwell clicking as a feature, and a field to set the dwell time before clicking.
When I set the circle size to 25 and the dwell time to 1000 milliseconds, it works perfectly with the communicator app, using eyecontrol only and no headmovement (head movement scanning works, but I prefer to not using it). 
Next step is to let my sister try it, and see if she is capable of controlling her eyes enough to use it.
I will release the new features soon, but right now it is a very quick an dirty code, so I need to clean it up a bit.
@schwabdidier : I have tried the GazeSpeaker very briefly, but found it a bit unstable, and not very easy to use. But the speaking functions in Communicator works perfect with PGM now, as well as a lot of other functions, so I can recommend it very much.

schwabdidier, 2018-02-11:
@Nbech : of course Communicator is better but it costs 600 euros. There are some other tools with a price > 500 euros. Mind express, grid,... I am currently trying to compare them before a such expense on a tool

nash8350, 2018-02-16:
Great just submit a pull request if you'd like to share your improvements @Nbech

nash8350, 2018-06-26:
Closing due to no update. Feel free to open a new issue if you have contributions.

closed 2018-06-26 by nash8350