PrecisionGazeMouse/PrecisionGazeMouse #13: Problems and a Suggestion

Eric-cirE, 2018-02-22:
I use a computer for my work but am suffering from arthritis (I'm only 20!) and am searching for a good way to go mouseless.  I use Voice Attack with 300+ custom commands for most things including clicking but I need to get rid of mousing with hands entirely.  I greatly hope PGM (or something like it) continues to be developed and becomes a perfect way to go 100% mouseless

3 Problems I found while using PGM (using Tobii Eye Tracker 4C and windows 10) :

1) Head rotation tracking stops *often* for no apparent reason but comes back after a few seconds.

2) The sensitivity level of the head tracking affects the mouse warping accuracy greatly.

3) Every time I've tried PGM after a short while my eye tracker stops working until I restart my computer, 4 times so far...

Suggestion : Add an option of manipulating the cursor location with your head like a joystick so if you turn your head a direction it will continue to move the cursor in that direction (faster the farther you turn) with an adjustable dead zone and top speed.

People, please help this guy develop PGM so you and others don't get RSI!

I believe this app has the potential to save my career if developed further. 

Thanks

trishume, 2018-02-23:
If you're also using voice stuff, and you want an actively developed head+gaze mousing + voice solution, you should check out https://talonvoice.com/

nash8350, 2018-02-24:
@Eric-cirE I continue to develop PGM but it meets most of my needs today. I'm a similar situation but with carpal tunnel. I've managed to go maybe 95% without the mouse. I still grab it a small portion of the time just out of convenience.

In regards to your problems:
1. Which tracker and movement mode are you using? Also, are you sure that the tracker is able to capture your head rotation? Sometimes my webcam loses tracking ability when I work in a dimly lit room, but I haven't had problems in a brightly lit room.
2. I need to know which tracker you're using since the sensitivity may be implemented differently.
3. Do you have the same problem with other eye tracking software, such as video games that use the 4C? Trying to figure out if you have a device or USB problem, whether it's an issue with the software.

Your suggestion: I like the idea of varying the displacement based on how far you rotate. @trishume suggested something similar on #9 . I personally haven't needed this but I'm open if someone wants to add it.

One problem with tying the cursor velocity to the rotation of your head is that when you turn to look at the right of left side of the screen you might not want the cursor to move. Even a small angle of difference outside the dead zone could result in the cursor moving continuously. At least with an absolute positioning system, the cursor will still remain relatively close. Is there a reason you prefer this joystick style?

trishume, 2018-02-24:
I've tried systems that do joystick style movement and they're extremely difficult to use, and have the problems @nash8350 lists. A better approach is the one I use in https://github.com/trishume/FusionMouse which is to use a good acceleration function so fast head movements fling the cursor further. That way you can reach the full screen while maintaining high accuracy.

Eric-cirE, 2018-02-26:
Thank you for the response.

1) This one is my problem.  The same thing was happening in the tobii eye tracking Unity demo stuff as well.  Now it's not doing anything head tracking at all!

2) ‎ I was using "EyeX only" mode in PGM with my Tobii 4C.

3) ‎ This problem might have something to do with my problem number 1.  I think I'm going to have to contact support.  x_x

I have not tried joystick style movement, so I don't know how annoying it is.  The reason I suggested it is because I often need to adjust value sliders, zoom, pan and orbit in 3D.  With PGM tilting my head to orbit, pan and zoom was very natural and cool except the current system only lets me go so far in any direction.

Thank you for the suggestions @trishume.  I'm a Windows 10 user though, so talonvoice would have to be a real good full solution for me to switch to Mac.

Tip for anyone who sees this that clicks by voice  :  Saying "toe" to click is much easier than saying "click" to click.  And "pip" for right click.

I'm now trying to develop my own eye tracking system in Unity.  I'm wanting it to work with Voice Attack in a special way.  Call me crazy.

nash8350, 2018-02-27:
Yeah bummer it sounds like a support issue for Tobii. I hope they can help you solve it! If you rule that out and still have problems let me know.

If you need to adjust the sensitivity more than the bar allows, you can customize it here https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/PrecisionPointers/EyeXPrecisionPointer.cs#L78

Yes I use custom commands like "kick" for clicking. I use caster which lets me build commands in Python and it supports programming by voice https://caster.readthedocs.io/en/latest/caster/doc/Intro/. Voice attack looks pretty cool too.

Eric-cirE, 2018-03-04:
I am confused. I uninstalled tobii eye tracking core version 2.11 (the latest) and install version 2.8, now head tracking works perfectly in the unity demo scenes! but nothing head tracking in PGM.  How is this possible?

nash8350, 2018-03-07:
The latest for me is 2.10.20. When I check for updates it does not offer a newer version. Both eye and head tracking work for me, but the head tracking is so noisy it's basically unusable. Not entirely sure that EyeX head tracking will give you precise control. It has to be pixel-accurate with <10 degrees of rotation. It's probably better for looking at broad parts of the screen rather than mouse movements. How accurate is it for you in the unity demo scene?

For precise control with head tracking, the eViacam or TrackIR offer much better accuracy in my experience. They don't add much in terms of CPU or cost either.

Eric-cirE, 2018-03-10:
When I checked for updates it didn't offer version 2.11.(~) for me either, but it's on their website.  I was on version 2.10.1(~), which for head tracking was literally only half working, it would toggle every 5 seconds between working and not working.  I think I'll try it again to see if it decides to work.  Either way I'll probably report these version problems to Tobii. Except I'm too busy making my eye and head tracking app while using version 2.8.  :)

As far as head tracking accuracy goes, those other systems you mentioned are surely better at it, being designed for the sole purpose of head tracking.  My Tobii 4C recognizes head angles of 10 degrees (maybe not too much more though) , but pixel accuracy is entirely dependent upon how sensitive you it set.  I plan on having different sensitivity levels for different tasks that are switched through voice control.

I'm getting closer, I've got Voice Attack to send commands to my program using "named pipes" (that wasn't fun). Next I need to find a way to stop it from controlling the mouse when I move the physical mouse...

I'm also planning on adding head position into the mix as well, might help me slide value sliders and who knows what else.  

This might be a dumb question but does the precision of eViacam or TrackIR encourage you to make tiny neck fatiguing motions?  I'm new to head tracking and my app is not to a usable state yet.  If I end up needing it I should be able to alter my app to utilize one of those systems if they support it.

Eric-cirE, 2018-03-10:
I did not mean to cross out any text in my previous post.  I did not know that those little wavy looking dash symbols would do that.  I meant to indicate that I didn't know the *exact* version because I'm not at my computer.  Live and learn.  *_*

nash8350, 2018-03-12:
Sounds like a fun project! Actually I find tiny motions much less fatiguing than large ones. I suspect the  effort required increases the further you go from a neutral position. It is also more fatiguing to hold an extreme angle for a long period of time or to make frequent switches between large angles. None of this is tiring after a few minutes but could be after 8 hours a day. That's why i created PGM to minimize head movement.

Eric-cirE, 2018-03-23:
Only time will tell but I got my eye tracker app working and about how I want it.

1) I checked for a new version of the tobii software a few days ago on Tobii's website and to my surprise version 2.13 was there. My program and PGM are now both working.

2) About the warping accuracy while using eyeX only mode in PGM, I understand the problem now after making my own.  It is adding the user's head rotation to the mouse after warping.  I made mine use the current head rotation as an offset  when warping the mouse so the mouse lands on the gaze point directly.

In my program I also have the mouse warp after being a small distance from the gaze point.  To the point blinking resets it. Both an advantage and a disadvantage.  I think adding a very small time threshold for the Gaze point to be in a vicinity before warping might solve this problem.  On that note I'll add this to my list of things to do.  :)

3) I won't be using PGM so I can't say if it messes my eye tracker up to the point of restarting my computer.

It might be fixed by the tobii update I've installed.

nash8350, 2018-06-26:
Cool glad you found a solution closing this issue

closed 2018-06-26 by nash8350