PrecisionGazeMouse/PrecisionGazeMouse #14: Eyetracker 4c & eViacam Prevents Pausing

Combateye, 2018-02-23:
I have an tobii eyetracker 4c that functions as expected in eyex only mode. When I use a webcam with eViacam enable and in eyex + eViacam mode (continuous), moving the mouse no longer pauses the eyetracking and the mouse pointer moves in unhelpful ways.

nash8350, 2018-02-23:
Interesting I have been using the on keypress mode so I hadn't encountered this one myself yet. I just tried it and I was able to reproduce the problem you described. 

I think the reason this happens is because I detect mouse movement by seeing if it changes without PGM moving it. That normally means the user moved the mouse. However, we cannot tell whether it came from eViacam or the user's mouse. If we paused on one we'd have to pause on the other. 

I'm not sure of a good way around this other than developing a closer integration with eViacam. I'm open to it if the developer of eViacam is interested.

The other way to pause the mouse would be to temporarily switch to on keypress mode. We could also add a pause button to the UI. Do you feel any of these is better?

Combateye, 2018-02-24:
Having a user configurable hotkey for pause seems like it would be the swiftest way to avoid the issue. Perhaps that's the best place to start?

nash8350, 2018-02-24:
That's a good idea! Might have some time to work on it next week. Feel free to submit a pull request though.

nash8350, 2018-02-28:
@Combateye I just released a new version with a pause hotkey in v1.07. I hope this can help you out! I'll close this issue for now, but please reopen it if you run into any problems.

closed 2018-02-28 by nash8350