PrecisionGazeMouse/PrecisionGazeMouse #15: Aplication does not work without Internet connection

Kaayl, 2018-07-29:
Hello, I download your application, everything works great - this app is what I was looking for.
However is there a way to run this app without internet connection? Some stand alone version perhaps?

At the moment application require network connection to be run.

Kaayl, 2018-07-30:
Hi, 
Thank you for answer.
I'm using launcher dowloaded from you webpage. Here is what I get after turning off WiFi and click on the lanuncher. Every time it tries to download some content. If I have access to the internet there is no issue, but without I get this: 
 
[image]

In details log I can find this:

> ERROR DETAILS
	Following errors were detected during this operation.
	* [7/30/2018 8:48:10 AM] System.Deployment.Application.DeploymentDownloadException (Unknown subtype)
		- Downloading http://precisiongazemouse.com/install/PrecisionGazeMouse.application did not succeed.
		- Source: System.Deployment
		- Stack trace:
			at System.Deployment.Application.SystemNetDownloader.DownloadSingleFile(DownloadQueueItem next)
			at System.Deployment.Application.SystemNetDownloader.DownloadAllFiles()
			at System.Deployment.Application.FileDownloader.Download(SubscriptionState subState, X509Certificate2 clientCertificate)
			at System.Deployment.Application.DownloadManager.DownloadManifestAsRawFile(Uri& sourceUri, String targetPath, IDownloadNotification notification, DownloadOptions options, ServerInformation& serverInformation)
			at System.Deployment.Application.DownloadManager.DownloadDeploymentManifestDirect(SubscriptionStore subStore, Uri& sourceUri, TempFile& tempFile, IDownloadNotification notification, DownloadOptions options, ServerInformation& serverInformation)
			at System.Deployment.Application.DownloadManager.FollowDeploymentProviderUri(SubscriptionStore subStore, AssemblyManifest& deployment, Uri& sourceUri, TempFile& tempFile, IDownloadNotification notification, DownloadOptions options)
			at System.Deployment.Application.DownloadManager.DownloadDeploymentManifestBypass(SubscriptionStore subStore, Uri& sourceUri, TempFile& tempFile, SubscriptionState& subState, IDownloadNotification notification, DownloadOptions options)
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivation(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl, Uri& deploymentUri)
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivationWithRetry(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl)
--- End of stack trace from previous location where exception was thrown ---
			at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivationWithRetry(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl)
			at System.Deployment.Application.ApplicationActivator.ActivateDeploymentWorker(Object state)
		--- Inner Exception ---
		System.Net.WebException
		- The remote name could not be resolved: 'precisiongazemouse.com'
		- Source: System
		- Stack trace:
			at System.Net.HttpWebRequest.GetResponse()
			at System.Deployment.Application.SystemNetDownloader.DownloadSingleFile(DownloadQueueItem next)

nash8350, 2018-07-31:
Oh that is only the first time you install it because it downloads it from precisiongazemouse.com. I suppose you do have internet since you were able to download the program? You don't need an Internet connection to use the program. I think if you installed it and then turned off your Wi-Fi you would see that it works fine. Does that work for you? If not, could you explain why you need it a different way?

Kaayl, 2018-08-01:
Ok, I now see my mistake. I though that this launcher was actual application. I didn't noticed that something was installed on computer. For future it would be nice if some shortcut was creadted on desktop after installation. :)

Thank you for help.
If I may I will open new topic with some question regarding configuration.

nash8350, 2018-08-07:
Ok great I'm glad you figured it out! It adds an item to the start menu but I'll keep your desktop icon idea in mind. Closing this issue.

nash8350, 2018-08-07:
Oh and yes feel free to open another issue if you have a question about configuration

closed 2018-08-07 by nash8350