PrecisionGazeMouse/PrecisionGazeMouse #16: Sensitivity

Kaayl, 2018-08-17:
Hello.

I noticed that mouse pointer moves when you are moving out side the gaze circe. **Is there an option or some parameter to change gaze circle size/radius?**

By that, I want to achieve that, pointer will move more frequently - and will be triggered by smaller eye movement.

Issue example:
[image]

If there are two icons inside Gaze circle with such big radius, then pointer will not move between icons.

nash8350, 2018-08-17:
Have you had a chance to figure out how to use the precision pointer yet? This will allow you to select smaller targets within the circle, such as one of the two icons. In my case, I use my head rotation for the precision pointer. I like to use eViacam for tracking my head rotation because it's accurate and I don't need to wear any headgear. It just uses a standard webcam to track my head position.

By having a smaller gaze circle, there is more chance that the pointer could move unexpectedly due to random noise in the eye tracker. However, if you are getting better eye tracking accuracy than I am it might make sense to reduce the size of the circle.

Kaayl, 2018-08-17:
I'm sorry, I'm not sure what you mean by "precision pointer". I don't know how to enable this option. Could you please advice how to use it?

Kaayl, 2018-08-17:
Also... If you could advice how to change gaze circle it would be most appriciated.
Is this size parameter located in some file? so I can just edit it.

nash8350, 2018-08-27:
Read the "how it works" section here https://precisiongazemouse.com/. One important thing to understand about the mouse is that there are two ways of moving the mouse. The first is that it can warp instantly to a location on the screen using eye gaze. For more precise control, you can use head tracking to place the cursor within a pixel of your target.

It may not be necessary to reduce the size of the tracking circle in order to click on the location you desire. That's because you should be able to use head tracking to move the cursor precisely. 

However, if you still prefer to reduce the size of the warp circle, you can do so by editing this line in the code: https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/WarpPointers/EyeXWarpPointer.cs#L21. You will have to build the program again. You can do so using the free Visual Studio community edition.

Kaayl, 2018-08-28:
The reason why I'm stubborn about changing gaze circle size instead of using head movement tracking is because my Daughter's condition not allow to use this method of tracking. 

By using relavively larger buttons/symbols for comunication she is able to point them quite precisely.

Thank you once again for your help. I will try to make size changes. I will back to you in couple of days to let you know how it works.

nash8350, 2018-08-30:
Oh that's cool I haven't thought about that use before but it makes sense! By the way, for full eye control Microsoft offers a free version integrated into Windows https://blogs.msdn.microsoft.com/accessibility/2017/08/01/from-hack-to-product-microsoft-empowers-people-with-eye-control-for-windows-10/. Of course, I would love you for you to use precision gaze mouse, I just want to share solutions I've heard of.

Kaayl, 2018-08-30:
Thank for hint. I belive I saw this Windows beta eye control plugin but it was still not enough satisfying for us.
If you are curiouse... we are using you program to control aplication called Eyefeel - produced by AssisTech from Poland.
There is EN language and 30days trial version for download if you want to take a look. Here is the link: https://assistech.eu/en/eyefeel-en/

nash8350, 2018-09-05:
This could become a configuration setting similar to #17. Will take some work. Marking it as help wanted for now.

nash8350, 2018-12-04:
Hey just wanted to let you know that I addressed this issue in version 1.11. If you download the newest version you can configure it in the Windows user settings. I didn't make it part of the UI because I'm not sure that everyone needs this feature. However, this makes it accessible to you without having to compile the code. 

To change the size of the warp threshold, first install and run the new version. Now open Windows Explorer and enter in the address bar %LocalAppData%\PrecisionGazeMouse. Next, find the newest folder and open user.config. Paste in the following:

```xml
<setting name="EyeXWarpThreshold" serializeAs="String">
    <value>500</value>
</setting>
```

Set the size of the circle to any size you like, in pixels. If you can't get it working just let me know! Closing for now but reopen if needed.

closed 2018-12-04 by nash8350

Kaayl, 2018-12-05:
Hi,
Thank you very much for your help. I will test latest update shortly. I really appreciate your help.

Kaayl, 2018-12-14:
Hi Nash,

I've tested your update. Setting new size of gaze treshold by changing value in config file is working. Thank you for that! 👍

nash8350, 2018-12-14:
Glad it works! If there's anything else I can do just let me know.