PrecisionGazeMouse/PrecisionGazeMouse #17: How to save (or set default) configuration.

Kaayl, 2018-08-17:
Every time when I open PrecisionGazeMouse I need to set desired option like for example: 

option Movement = Continuous

Is there a way to set this option as default?

nash8350, 2018-08-17:
This is a great suggestion! I haven't had a chance to implement this yet because I just use the default mode. However if you prefer a different mode I could see this being an improvement.

Kaayl, 2018-08-17:
If there are any vars in some file that I can set with right values (for exmaple for default Continuous Movement option) thats also fine for me, to just edit some config file - no need for fancy "save" button :)

Just point me in right direction - in which file it can be changed.

nash8350, 2018-08-27:
Currently, the defaults are hardcoded so there is not a configuration file where you can change that. The change them in the code you can edit this file https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/PrecisionGazeMouseForm.cs#L28

I would like to add a configuration file in the future so that we can save the settings. I have marked this issue as "help wanted". If you would like to contribute this improvement, please do so. Otherwise I can try to do it when I get a chance.

nash8350, 2018-11-23:
Finally got a chance to add this feature in v1.10. Install the latest to get it.

closed 2018-11-23 by nash8350