PrecisionGazeMouse/PrecisionGazeMouse #19: Hardware advice

JoasNL, 2018-09-27:
Hello,

I stumbled across your software and I think it's awesome! I would like to start using it, but I need some advice on which mode is the most precise and what hardware I require for that. Can you help me with that? I do wear glasses sometimes, is that not possible with this technique? 

I have a two monitor setup, do you think that would work? Not a 100% necessary, but it would be nice.
 
If you have any affiliate links from amazon.de or amazon.co.uk I'll be happy to use them.

Found some small typos in your text: 

'means it takes longer **to** click'   
'Keeping your eyes focu**s**ed '

Thanks for your time!

nash8350, 2018-10-08:
I wear glasses too and it works fine with eViacam and EyeX. EyeX only supports a single monitor I believe. I use an EyeX 4C and Logitech C922 camera and they work well. I don't have any affiliate links but if you use smile.amazon.com it will go to a charity. Thanks for the note about the typos I just fixed them.

nash8350, 2018-10-15:
Closing this since I hope I addressed your question. Reopen if needed.

closed 2018-10-15 by nash8350