PrecisionGazeMouse/PrecisionGazeMouse #20: TrackIR 5 use as cursor with PGM

ealfiey, 2018-09-27:
Hello, I have ALS, no use of my hands anymore. I'd like to use PGM as mouse/cursor.
I already had a Tracker 4C for use with PGM and now I have recently bought a 2nd hand TrackIR 5.
I installed it with software version 5.4.2.
I can't see how to set it into mouse/cursor mode. 
Also PGM indicates that it is paused.
Please help!
regards, 
Evert, Netherlands

nash8350, 2018-10-08:
Hi @ealfiey sorry to hear about your ALS and I hope PGM can help. Which movement mode have you selected? Can you confirm the pause key is not activated? Also, it may pause if other devices are moving the mouse cursor. Can you confirm PGM is the only one?

ealfiey, 2018-10-24:
hello Nash,

I have a new PC built. I have to arrange some reasonable smart person to
help me with the transfer.
If the problem persists with the new PC, I will contact you again.
Thank you for your help.

kind regards,
Evert (Ealfey)