PrecisionGazeMouse/PrecisionGazeMouse #21: Using Precision Gaze Mouse with dual monitors

nickjw2137, 2018-11-05:
Hello, 

First of all, I would like to say thank you so much for creating this software and sharing it with people for free! 

I am wondering if I have the Tobii eye tracker and webcam set up on one monitor, will it track the mouse onto a second monitor as well?

Also, is there a certain webcam that you recommend using? Would the camera that is built into my laptop work?

Thank you for all of your help.

nash8350, 2018-11-07:
Hey thanks glad to have people use it! AFAIK the Tobii EyeX only supports a single monitor and that's what I use. I think eViacam doesn't have this limitation, but you'll have to rotate your head further to cross to the second screen. I think you can use it with the camera built into your laptop. It works fine for me when I'm at home. I do have a nicer camera at the office and that is a Logitech C922.

nickjw2137, 2018-11-07:
Cool thanks for all the info! I don't think the one monitor limitation will be much of a problem since I usually just mouse on my main screen anyway. 

Thanks again!

closed 2018-11-09 by nash8350

ethangoldstein, 2019-07-02:
If there was a second tobii4cc for the second monitor, would it be feasible to support 2 monitors?

nash8350, 2019-08-15:
@ethangoldstein I did a quick search and it looks like Tobii still doesn't support multiple screens https://help.tobii.com/hc/en-us/articles/209529429-Can-I-use-multiple-screens-