PrecisionGazeMouse/PrecisionGazeMouse #24: Startup Behavior on Continuous Mode

eladnarra, 2019-05-13:
Hi! Firstly, I love this program. It's exactly what I was looking for when I started trying to use eye tracking for work when my RSI acts ups. All the other programs I tried were too slow and/or too inaccurate for me. Thanks for putting it online for folks to use!

My setup: this program w/ eViacam, Fragpedals, and a tobii eye tracker. I'm in Windows 10 and use continuous mode. I set both programs to run on startup.

My small issue is with start up behavior. For some reason even though it defaults to Continuous mode, the eye warp function doesn't work. I have to click OK on the dialog box and then click the Continuous radio button (even though it's already selected), then click OK again. It's a minor thing, but it means I have to use my mouse, which some people might not be able to do at all. 

It would also be neat, but less necessary, if one could permanently get rid of the dialog box that always comes up when you're using eViacam and switching modes. (A "don't show this again" option?)

eladnarra, 2019-05-27:
An update: I actually read the text in the initial dialog box more fully, and when the program starts up it displays the warning/notification for "On Key Press" mode. (Make sure eViacam is disabled.) So it seems like the program defaults to key press mode despite showing the radio button for continuous mode as selected. Is there a way to make it start in Continuous mode? Thanks!

caspark, 2019-06-16:
I have a fix for this and a couple related issues in PR #26 .

eladnarra, 2019-07-13:
Cool, thanks. Now I just need to figure out how to install from github, I guess... (previously did from the website, which is now down anyway).

closed 2019-07-13 by eladnarra