PrecisionGazeMouse/PrecisionGazeMouse #25: Error on start "Unhandled domain exception"

lexxish, 2019-05-29:
On opening setup.exe for v1.11, a popup comes up containing `Unhandled domain exception: the type initializer for 'PrecisionGazeMouse.Program' threw an exception` and the setup fails.

On Windows 10 with Tobii 4c connected and .Net Framework 4.7.03190.

lexxish, 2019-05-31:
@nash8350 any idea why I'm getting this error?

lexxish, 2019-06-16:
This happens when building from source too. I believe this may be a bug in the Tobii SDK. Here is the stack trace:

> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_32\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. 
> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\code\PrecisionGazeMouse\PrecisionGazeMouse\bin\x86\Debug\PrecisionGazeMouse.exe'. Symbols loaded.
> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: PrecisionGazeMouse.exe): Loaded 'C:\code\PrecisionGazeMouse\PrecisionGazeMouse\bin\x86\Debug\Tobii.Interaction.Net.dll'. 
> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: PrecisionGazeMouse.exe): Loaded 'C:\code\PrecisionGazeMouse\PrecisionGazeMouse\bin\x86\Debug\Tobii.Interaction.Model.dll'. 
> Exception thrown: 'System.BadImageFormatException' in Tobii.Interaction.Model.dll
> An unhandled exception of type 'System.TypeInitializationException' occurred in Unknown Module.
> The type initializer for 'PrecisionGazeMouse.Program' threw an exception.

> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: PrecisionGazeMouse.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll'. 
> 'PrecisionGazeMouse.exe' (CLR v4.0.30319: PrecisionGazeMouse.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll'. 
> Unhandled Exception: System.TypeInitializationException: The type initializer for 'PrecisionGazeMouse.Program' threw an exception. ---> **System.BadImageFormatException: An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)**
>    at Tobii.Interaction.Client.Interop.EyeX.IsEyeXInitialized(Int32& pInitialized)
>    at Tobii.Interaction.Model.Environment.get_IsInitialized()
>    at Tobii.Interaction.Host..ctor(String name)
>    at PrecisionGazeMouse.Program..cctor() in C:\code\PrecisionGazeMouse\PrecisionGazeMouse\Program.cs:line 17
>    --- End of inner exception stack trace ---
>    at PrecisionGazeMouse.Program.Main()
> The program '[14120] PrecisionGazeMouse.exe' has exited with code 0 (0x0).

lexxish, 2019-06-17:
I did get this working. It seemed to have to do with the version of .net I was using. I retargeted for 4.5.2 and built from source.

nash8350, 2019-09-09:
I just updated the source for 4.7.2. I didn't make a release for it since there is no functionality change, but reopen this issue if you still get this problem with the latest source.

closed 2019-09-09 by nash8350