PrecisionGazeMouse/PrecisionGazeMouse #27: Help compiling?

eladnarra, 2019-07-13:
I have a new computer and I can't install from an old application file because the website is down. I don't exactly expect anyone to be able to help me, but I thought I might as well ask: is there any documentation on how to compile the program, or someone willing to explain how to do it?

I am pretty confident with computers, but don't have any experience in this area. I'm more at a level of "dabbled in python and javascript, took a class that involved using linux terminal, knows how to google." So I can follow instructions, or maybe even figure it out with some links, but don't know enough to know what to google in the first place.

Thanks in advance, and if this is too time-consuming of an ask, I understand. (Although if the website stays down, I'm probably not the only person who will want to know how to install without the website/application file.)

nash8350, 2019-08-15:
@eladnarra Hey sorry I am working on renewing my domain and hope to have it done soon. Then you'll be able to install from the website again. 

Nevertheless if you want to run it from the source code you can. Right now it builds in Visual Studio 2017. Just open it up, hit f5 to build it, and then double click the exe file that it generates in the "bin' folder. There's no need to install it, unless you want the start menu entry.

nash8350, 2019-09-06:
@eladnarra I finally got the new version released you can install it directly from https://precisiongazemouse.org/install/publish.htm. Let me know if you have any problems.

closed 2019-09-06 by nash8350