PrecisionGazeMouse/PrecisionGazeMouse #29: Is this still supported?

edmun777, 2019-09-03:
I downloaded the .exe file from TechSpot and got an error message immediately upon attempting to install.  Tried to access developer website, and keep getting an IP address error.  Cannot find installation guide/help anywhere!  Is this software still supported as anything other than source code?  Can anybody help me with this?

Technetium1, 2019-09-05:
@edmun777 You should be downloading from the releases not techspot https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/releases

nash8350, 2019-09-06:
Yes it's still supported. I had to switch the installation to a new domain as explained in #31. If you go here you should be able to get the latest version https://precisiongazemouse.org/install/publish.htm.

I don't have any relationship with TechSpot I think they are just rehosting it so make sure you get 1.12 which is the latest. If you have any problems let me know.

closed 2019-09-06 by nash8350