PrecisionGazeMouse/PrecisionGazeMouse #30: Suport for WebCam Eye-Tracker

szydej, 2019-09-06:
Have you ever thought about supporting WebCam based Eye-Tracker? 
GazeFlowAPI  https://github.com/szydej/GazeFlowAPI
GazePointer https://sourceforge.net/projects/gazepointer/

nash8350, 2019-09-06:
@szydej This looks pretty cool thanks for sharing it! We have head tracking through Enable Viacam but not gaze tracking through a webcam yet. More people have access to a webcam over an EyeX so this might broaden the reach. I see that are you the author would you potentially be able to help integrate it?

szydej, 2019-09-07:
GazeFlowAPI is really easy to integrate (check out source code of sample client https://github.com/szydej/GazeFlowAPI/blob/master/C%23/GazeFlowAPI.cs), let me now if you have any questions.

nash8350, 2019-09-09:
Hmm ok I'll get it out when I get time. I'm marking this help wanted in case anyone else gets to it before me.

nash8350, 2019-09-16:
@szydej I gave gazepointer a shot. I love the convenience of using just a webcam. However, for me the accuracy was lower than EyeX. With EyeX the pointer is usually within 1 inch of where I am looking. With gazepointer it was more like 2-3 inches. I imagine many factors can affect the accuracy and some people get better results? I have a 1080p camera and maybe it would be better with a 4k camera or if I wore contacts instead of glasses? As it is, I don't see myself switching but I'm happy to leave this issue open for people who prefer this approach.

szydej, 2019-09-25:
@nash8350  
You can try using it for a while. Every time you click anywhere on the screen your eyesight accuracy continue improve.

OpenlabWA, 2020-03-13:
@nash8350 I've been interested in researching a viable mouse-free system for a while now, then came across your youtube video yesterday. 

I'd not heard of the out of the box tools for tracking that you're utilising, my idea was to use 1 or 2 1080p (maybe 4k?) webcams to track head and eye position, I really love what you've got so far and I might get my a into g and get some cameras to begin with. 

I didn't dig too deep, but, it appeared that there were no blink detection properties provided by the hardware, disappointing, I'm keen to implement left click and right click based on left or right blink, and potentially some blink gestures, such as zoom, click and hold or something along those lines. Given the great accuracy that you're getting, if a couple of 'cheap' 1080p webcams could be used in a stereoscopic mode and provide a similar output, I reckon it would really take the initial sting out the hardware purchase side of things.... What do you think?

Cheers Muzz.

nash8350, 2021-03-12:
@OpenlabWA hey missed your comment. If you get blink tracking working let me know. It looks like Tobii and eViacam don't support that yet. I'd be surprised if you couldn't find something with blink tracking out there. It looks like dwell clicking is more common though. I happen to use a USB button. You can also click by voice.