PrecisionGazeMouse/PrecisionGazeMouse #32: Hiding eViacam Interface

edmun777, 2019-09-10:
When integrating Precision Gaze Mouse with eViacam, is there a way to hide the eViacam click bar after setup?  I am planning to use voice clicking, and I would like to hide the bar so it does distract from or cover other things I'm working on.

Brotie, 2019-09-28:
Open up eViacam go to settings or config. Under Click make sure "dwell clicking" isn't selected that will make sure that the "click window" option won't be selectable. And that's all you can use whatever other method to click without unnecessary toolbars.

nash8350, 2019-10-16:
Great thanks for helping @Brotie. I'm closing this for now but if that doesn't help please reopen this issue @edmun777

closed 2019-10-16 by nash8350