PrecisionGazeMouse/PrecisionGazeMouse #35: High CPU useage

ethangoldstein, 2019-12-15:
Hi,

When I use the software I get a around 20% of my desktop Quadcore i5 being used up. Is this what other users are experiencing?

Thanks.

nash8350, 2019-12-17:
I've also got a quad core i5 and I see about 1% CPU usage for Precision Gaze Mouse and 5% for Enable Viacam when I'm actively using it. Are you using the original EyeX or 4C? The 4C has an onboard ASIC chip to process eye gaze with very low CPU. Also, make sure the "Show Warp Bar" and "Show gaze tracker" checkboxes are turned off because they each use extra CPU.

caspark, 2019-12-30:
I haven't got an EyeX to compare with, but for what it's worth I pretty regularly see high CPU usage from Tobii's 4 processes when using my 4C, which often persists even after PrecisionGaze is turned off or exited.

I haven't tried to prove it but I can't help but feel that Tobii's software is doing needless work.. possibly related to all the other not-so-helpful features the software provides, such as choosing alt-tab targets by gaze (or highlighting its tray pop-up menu via gaze).

EViacam is definitely more CPU hungry though!

ethangoldstein, 2020-01-01:
I am not sure why it still consumes a high amount of CPU for me. I started using the talon eye tracking and now the CPU usage is extremely low. I also have a strong feeling that the Tobii driver is very wasteful.

nash8350, 2020-01-05:
Interesting I know the Talon eye tracker has a custom driver the author wrote himself so maybe it bypasses some issue with the Tobii driver. That still doesn't explain why CPU usage is low for me but not you. Perhaps there is some system level difference? For what it's worth I do have most of the optional interaction features turned off. 

If the problem is with PrecisionGazeMouse and not Tobii, you could run it in the debugger and see how much CPU is spent in various functions https://docs.microsoft.com/en-us/visualstudio/profiling/profiling-feature-tour?view=vs-2019. That requires some development experience though, and spare time to run the test.

nash8350, 2020-02-16:
Closing since it seems this requires more information to fully debug. If you do find where the CPU is being used please reopen.

closed 2020-02-16 by nash8350