PrecisionGazeMouse/PrecisionGazeMouse #36: Eye movement cancels Dragon commands

edmun777, 2019-12-15:
For some reason, I cannot seem to get Dragon to voice click on items while EyeX is active.  It appears any movement of the eye will displace clicker before it clicks.  For example, saying “save” or “click save” does not register unless I hold completely still.  Any workaround for this?

nash8350, 2019-12-18:
Interesting is the problem that Dragon has too much lag to recognize and execute your command, and in the meantime you have looked elsewhere? Or that your EyeX gaze tracker is jumping outside the warp boundary and not staying in the head tracking mode long enough for Dragon to execute? Or do you find it physically difficult to hold your eyes or head still while Dragon executes?

Also, what Tracker and Movement mode do you have selected and what is your sensitivity set to?

edmun777, 2019-12-18:
I had it on TrackIR and eyetracker 4c, I think.  I have since switched to
eviacam with 4c, and it seems to work well so far.  It seemed mostly to be
an issue with saving in DragonPad, so I will check that with the new
settings tonight and let you know if that has fixed it.  Was not having
problems in other windows apps, which is a bit weird.  So not sure if the
TrackIR was “sliding” a bit, or if it’s just specific to DragonPad.

nash8350, 2020-01-05:
@edmun777 ok good to know you found a working solution. If you do figure out that other problem feel free to post details here and reopen. In the meantime, I'll close this issue.

closed 2020-01-05 by nash8350