PrecisionGazeMouse/PrecisionGazeMouse #37: Open sourcing software based on apache 2.0 license

rishikapadia7, 2019-12-18:
Hi Nash,

I used PrecisionGazeMouse as a starting point for making a fully functional mouse replacement comparable with Tobii Dynavox. I was not successful in commercializing it, hence, would like to open-source it. I have searched in-depth on how I can be compliant with the Apache 2.0 license and can't figure it out.
Could you please give me directions on what I should include when publishing my derivative work on github so that I am not in violation to your project?

nash8350, 2020-01-05:
@rishikapadia7 wow that's cool to hear! I'd love to see can you share a link? Also, I'm curious what you changed/added? 

I just picked the apache license because I wanted to make it open source without limiting the ability for others to commercialize it. It'd be nice to have attribution in your README or somewhere else. I'm not going to picky about it.

rishikapadia7, 2020-01-10:
Hey @nash8350, everything you'd like to see is right here:
https://github.com/rishikapadia7/AvraLabsEyeControl

I have the README as well as a credits section on my wiki :)

Definitely check it out, your project helped as a great stepping stone for working with the Tobii tracker, and now we have a full-out mouse and keyboard replacement :)

nash8350, 2020-01-17:
@rishikapadia7 great work! I played with Optikey but the mouse wasn't too good. This is a great improvement. Credits look good. A link always helps for SEO but thats up to you.

closed 2020-02-16 by nash8350