PrecisionGazeMouse/PrecisionGazeMouse #40: Head rotation stays at (0, 0)

brummel2, 2020-01-21:
Laptop:
Tracker Mode: Trackir Only
Head Rotation didnt get the coordinates until i found out PGM must run as admin...
All works fine now

Desktop:
Same issue as laptop, only the same sollution doesnt work.
TrackIr and PGM both run as admin, nothing there, coordinates stay at 0, 0

Laptop and Desktop both run on windows 10.

Please let me know what else i can try?

Thanks alot!

Ps, im gonna show you a nice video on how i use
PGM, combined with Trackir, Tobii 4c and voicemacro to play 9 tables of online poker soon

brummel2, 2020-01-21:
After some restarts and some more playing with admin rights i got it to work :)

nash8350, 2020-02-16:
Sorry for the slow response I had a busy month. I'm glad you got it working! Were there any particular permissions you had to enable that I could share with others?

nash8350, 2020-03-28:
Closing this due to no activity. If you have any information to add on permissions please reopen.

closed 2020-03-28 by nash8350