PrecisionGazeMouse/PrecisionGazeMouse #41: Cursor warps for no reason

edmun777, 2020-01-24:
Hi,
Lately I have been having issues with the cursor occasionally warping short distances without any eye movement (using eViacam and EyeTracker 4C).  It seems to jump just outside the warp radius, sometimes several times in a row--I will be looking at an icon, but before I can click, the cursor will jump off it; I will look away and look back to reset the cursor, then it will jump off (often in exactly the same direction) again, even though me eyes have not moved and I am still staring intently at the thing i am trying to click.  The issue seems to happen more frequently closer to the edges of the screen, so it may have something to do with hardware limitations of the EyeTracker (although my monitor is within the supported range).  I tried re-calibrating the eye tracker, and that seemed to fix the issue for about five minutes before it started happening again.  Any ideas on why this may be happening or what I can do about it?  It does not happen all the time, and I can usually click what I need on at least the second or third try, but it would be nice to be able to get things the first time.  Like I said, closer the middle of the screen, this almost never happens.

caspark, 2020-01-25:
When you turn on the debug overlay that shows you your gaze position, does that little circle move outside of the big circle? (That's the trigger for the warp.)

I have this happen sometimes (at the corners, using a 30" screen), but it's always explainable by the gaze point moving outside the big circle. I've mostly gotten used to it and work around it by moving windows around using shortcuts, but maybe it'd be a worthwhile change/option to make the warp threshold larger at the edges of the screen.

nash8350, 2020-02-16:
Yes @caspark is right the debug overlay helps here. The EyeX is less accurate at the edges of the screen. You can verify this in the calibration mode. Also make sure EyeX has big white dots instead of small grey ones for your eyes. The accuracy goes down when you sit too far from the screen. Adjusting the warp threshold can help too. I set mine high enough that it doesn't randomly warp, and I use my head rotation to make up for the gap around the corners. That's easier for me than retrying 2-3 times. Closing because that's all I can think of right now. If you find a good technique feel free to share  it though.

closed 2020-02-16 by nash8350