PrecisionGazeMouse/PrecisionGazeMouse #42: Application silently fails to launch

cybairfly, 2020-02-23:
Hello, first of all thank you for publishing this!

I have an issue with the application not launching at all after installation. Running the setup just silently places a link in the start menu without any proper installation process and launching the application afterwards results in the app briefly appearing in tasks (32bit) and then disappearing silently without any apparent action. Platform is 64bit Win10 on Asus Zenbook Pro.

Thank you in advance for any kind of advice.

caspark, 2020-02-23:
One of the first things PrecisionGaze does is check whether it's running as administrator, and if it isn't, it [attempts to relaunch itself as administrator](https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/Program.cs#L110). In theory if this doesn't work, the error is handled and the app should still run, but in practice perhaps this error handling isn't working?

You could see if it works when you right click on the PrecisionGaze executable and `Run as administrator` manually. You should be able to find the PrecisionGaze executable by searching for a file called `PrecisionGazeMouse.exe` in `C:\Users\YourUsername\AppData\Local\Apps`.

cybairfly, 2020-02-24:
Thank you for the speedy response! Actually looking for the executable I discovered a number of crash dumps corresponding to the number of attempted runs and I will attach one of them in case it would be of any use. Running as administrator doesn't help it. Another things I noticed is the files are nested in strange location, e.g. `C:\Users\Tobey\AppData\Local\Apps\2.0\ABY39EQR.BM9\G16E4AH4.Y5Y\prec..tion_0000000000000000_0001.000c_fe87e4cbef512a87`

cybairfly, 2020-02-24:
Crash dump file: https://file.io/DMU7GL

caspark, 2020-02-25:
Oh to be clear, I'm not the main developer (that's @nash8350 !) - I just contribute a patch or two and try to be helpful sometimes.

The oddly named directories are part of the ClickOnce installer used as far as I know.

I'll admit I don't know much about reading dumps, but the only thing I can get out of that minidump is:

```
System.TypeInitializationException
  HResult=0x80131534
  Message=The type initializer for 'PrecisionGazeMouse.Program' threw an exception.
  Source=<Cannot evaluate the exception source>
  StackTrace:
<Cannot evaluate the exception stack trace>

Inner Exception 1:
DllNotFoundException: Unable to load DLL 'Tobii.EyeX.Client.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
```

For me, `Tobii.EyeX.Client.dll` is in the same directory as the app's executable itself (and signed by Tobii with a valid signature). If you have the same, then that's about the limit of my C# debugging.

You could try cloning the repo and opening then debugging it in Visual Studio 2017 itself (just make sure to run VS as administrator or you might not actually be able to do any step debugging) or wait for @nash8350 to chime in.

cybairfly, 2020-03-01:
I see, in that case thank you to both you and @nash8350 !
I checked and yes I have the same signed .dll in the same folder as well.
I'm afraid C# debugging is beyond my skills but might give it a shot. 
Any further feedback from you guys would be appreciated.

nash8350, 2020-03-11:
Hey @cybairfly thanks for trying the app and sorry you ran into this error. Everything @caspark said is correct thanks for helping out! Unfortunately, the dump you made expired and it gives me a 404 error. 

Just to confirm do you have Tobii EyeX installed on your computer and it's running successfully? It almost looks like there is an error in the Tobii DLL that is preventing it from loading, but its hard to say without more info. I suspect it gives the same error message even when there is a error encountered while loading the DLL. That is unmanaged and closed-source code so its trickier to debug problems there.

cybairfly, 2020-03-21:
Hey @nash8350 thank you for releasing the app in the first place! My pleasure to help and try it out.

I installed the app without reading instructions and didn't have anything else installed actually. I don't have any Tobii device so I went for the eViacam option instead. Installed it, set it up, enabled hotkey for activation (F11) and then launched your app. Sadly, even when launched as admin it just keeps terminating silently nevertheless. Could UAC have any effect?

nash8350, 2020-03-28:
@cybairfly Ok well if you don't have a Tobii device installed, that explains why the DLL doesn't load. It shouldn't crash so that's something I need to fix.

I'm curious what is the reason you want to use it without a Tobii device? Right now that is the only eye tracker supported. If you wanted to use eViacam as a mouse, you can use the software directly without Precision Gaze Mouse. You only get a benefit when you use it with an eye tracking device.

cybairfly, 2020-05-03:
@nash8350 Apologies, my bad. I didn't realize a Tobii device was required because from reading the website it sounded to me like eViacam was an alternative to such device... as I don't own any, makes sense that the DLL isn't working. Thank you for the explanation and sorry about wasting your time.

nash8350, 2020-05-20:
Ok thanks I updated the website to hopefully make it a little more obvious. I'll leave the issue open for now as a reminder to test it without EyeX installed.

nash8350, 2020-11-07:
I added a better error message when tobii is not installed to v1.13. Hopefully if anyone runs into this again it will help them.

closed 2020-11-07 by nash8350