PrecisionGazeMouse/PrecisionGazeMouse #44: Support for new user

SaddamBInSyed, 2020-05-11:
hi @nash8350 

thanks for your work.

I have eviacam in my pc and working fine.

i come across this repo and i would like to know what special feature does this repo have over the eviacam?

i run this repo exe but no luck ( no camera connecting )
if any document to follow for new user please advise.

caspark, 2020-05-12:
PrecisionGaze allows eye tracking to be used for quick movements across big swathes of the screen while still allowing eviacam (or alternatives, like TrackIR) to be used for small cursor movements. If you do not have a Tobii eyetracker and have no plans on getting one, you can ignore this app. Otherwise, read the website linked from the github repo description for an introduction to the features.

SaddamBInSyed, 2020-05-13:
@caspark 
thanks for the info.  at present no plan to buy Tobii cameras.
i thought it will simply will work with logitec webcam like eviacam.

thanks again

closed 2020-05-13 by SaddamBInSyed