PrecisionGazeMouse/PrecisionGazeMouse #45: Tobii Eye Tracker 5

kb100, 2020-06-21:
It looks like Tobii has just released the Eye Tracker 5. I am interested in purchasing either the 4c or 5. Do you know or expect that the Eye Tracker 5 is compatible with PrecisionGazeMouse?

kb100, 2020-06-26:
Closing this issue, yes it works with the Eye Tracker 5.

closed 2020-06-26 by kb100

nash8350, 2020-06-27:
@kb100 thanks for testing it out! Apparently the 5 has better head tracking. I'm curious if it is accurate enough for use? There is an EyeX mode you can use to try it out.

kb100, 2020-07-15:
> @kb100 thanks for testing it out! Apparently the 5 has better head tracking. I'm curious if it is accurate enough for use? There is an EyeX mode you can use to try it out.

I just compared vs eviacam and i think it has the potential for use but it would need a bit of work. Namely, it needs smoothing and a minimum motion threshold, and it should use relative-to-last jump rotation coordinates. Currently if your head is both looking and facing the left side of the screen, then the mouse will jump to the left of where you are looking. I propose to store the rotation coordinates when the mouse jumps and use the current minus last jump coordinates. I think that it would be competitive with eviacam if those things were sorted out.

Xaklse, 2020-08-14:
Please check @nash8350: https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/issues/47#issuecomment-674199081