PrecisionGazeMouse/PrecisionGazeMouse #47: Tobii 5

henrik-dreyer, 2020-07-11:
Hi!

Does it work with Tobii 5?

joewilbert, 2020-07-16:
It's working for me with Tobii Eyetracker 5 on Windows 10.  Thanks to @nash8350 for making this!  The Eyetracker 5 itself has been a bit quirky for me, though.  Good accuracy, but was a pain to install, occasionally stops working, and doesn't currently work with Windows Eye Control (customer support said they're working on a solution but no ETA).

Xaklse, 2020-08-14:
The mouse pointer warps correctly, but head rotation doesn't work properly, I had to disable the 'precision' part with zero sensitivity.
I took a look to the code with Visual Studio... but I can't understand the use of some undocumented 'magic numbers', and logging doesn't seem to work (I'm not a .NET coder).

kb100, 2020-08-15:
Yes it works for me as well, but the head tracking is not as good as eviacam by a long shot. Code cleanup would be great, magic numbers are very hard to get over.

nash8350, 2020-10-26:
I added Tobii 5 to the user manual. I'm leaving this open as a reminder to label the magic numbers.

nash8350, 2020-11-07:
Thanks for the feedback on the magic numbers @kb100 and @Xaklse I added some better documentation for them here https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/commit/1aba3cbc7fd036a3874e88aa3b0b6beccaabe63b. Let me know if im missing anything.

closed 2020-11-15 by nash8350