PrecisionGazeMouse/PrecisionGazeMouse #49: How to Donate?

TubaMT, 2020-08-27:
Thank you!! Thank you!! Thank you!! Thank you!! Thank you!!

I know this probably isn't the right place to post this but I did not know where else to ask where I can donate.

This is the best software I have found to control a mouse without using hands. I would love to donate money but I am not seeing on any of your pages how I would go about doing that. Let me know how I can support you. You've saved me hours of time and probably thousands of dollars trying to find a eye tracking solution.

PS. I've tried, Smyle Mouse, CameraMouse, GazeRecorder, Project Iris, OptiKey, Tobii Dynavox GazePoint Free, Windows Eye Control, TrackIR Mouse, FaceTrackNoIR, eViaCam and maybe a few others I forgot the name of. Your software is the best of the bunch. The only other option I have found that doesn't even seem to compare, would be to get the AblePoint TrackerPro2 or the Orin HeadMouse Nano that only track head movements. Your software's ability to be able to combine both head tracking AND eye tracking is the game changer. Also the ability to not have click on dwell is super helpful too. Using the eye tracker by itself is way too jittery, your software solves this problem very well.

I currently use Precision Gaze Mouse with a Tobii Eye Tracker 5, eViacam (Logitch C920 Pro), and a combination of VoiceAttack, Windows Dictation, and foot pedals for commands (such as left click, etc). I've tried Precision Gaze Mouse with a Tobii Eye Tracker and a TrackIR but it doesn't seem to work as well as eViacam does. Could be a conflict in the IR reflectivity. I am pretty much completely able to use my computer hands free.

Thank you so much for this excellent software!

steven-foster, 2020-09-10:
I concur.
Also, which foot pedals are you using out of curiosity?

TubaMT, 2020-09-11:
We use a combination of: 

-This Lemo 3 pedal switch 1640835 (https://www.amazon.com/gp/product/B00B3PV47O/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). There are a bunch of these types of switches. Just search Amazon for USB Single, Double, or Triple Foot Switch. I have heard the Frag Pedals are also good (https://www.gamingmouse.com/).

-The Xbox Adaptive Controller (https://www.xbox.com/en-US/accessories/controllers/xbox-adaptive-controller) with the Logitech Adaptive Gaming Kit (https://www.microsoft.com/en-US/p/logitech-adaptive-gaming-kit-for-xbox-adaptive-controller/936ndvcjzp78?activetab=pivot%3aoverviewtab). We use these with a program called reWASD (https://www.rewasd.com/) to remap the buttons.

-Ablenet BIGtrack 2 (https://www.ablenetinc.com/bigtrack-2) and Kensington SlimBlade (https://www.bestbuy.com/site/kensington-slimblade-trackball-graphite-ruby-red/9765259.p?skuId=9765259) and 3dRudder (https://www.3drudder.com/) as a foot mouse.

-Apple Magic Trackpad 2 as a foot mouse as well. We did not like this as much because it is too easy to left click with your foot and it is harder to control than a trackball.

-Any USB desktop type of mic like the JLab Audio TALK GO (https://www.bestbuy.com/site/jlab-audio-talk-go-plug-play-microphone/6417452.p?skuId=6417452) to be used with VoiceAttack.

Thankfully there are a lot of options.

On a day-to-day basis, we use the Xbox Adaptive Controller with the Logitech Adaptive Gaming Kit with reWASD and the Lemo 3 pedal switch. We use the Ablenet BIGtrack 2 with our foot if we need a normal mouse or if we are tired from the eye tracker and head tracker. I have tried the other items mentioned above and they work also work well. It's all preference though.

Using VoiceAttack in combination with the foot pedals is a lifesaver. We don't even have to use the foot pedals if we don't want to. We have commands like "Scroll Up", "Left Click", "Dictation", etc. I even set it up to disable and enable with my voice the PrecisionGazeMouse, eViaCam, and Tobii Eye Tracking software and services so I can turn it on and off very easily. It is a super easy to use program with no programming skills needed.

I hope this helps you and anybody else looking for information on hands-free computer navigation. It took me a couple months to figure it all out, but I think we finally landed on a good solution for us. Let me know if you have any other questions and I'll try to answer them :)

nash8350, 2020-10-18:
Wow thanks so much for your encouragement! Similarly, I tried a half dozen other solutions but I wanted something better.  I'm glad others are using it and benefiting from it. That's why I shared it here, to help others. Thanks to this mouse and my voice recognition I have a pretty good job so I don't need donations.

I'd say please donate your time to help others discover Precision Gaze Mouse. I don't think it's well known in the community yet. Telling friends is a great start. Even better would be writing a blog about your experience. Bonus points if you write an article in a publication or mailing list for people with disabilities or injuries. Imagine how many people could benefit if they only knew this existed!

TubaMT, 2020-12-28:
Yes I was surprised how hard it was to find out solutions for hands-free computing. I thought it would be farther along than it currently is.

I found out about your program through the Tobii Eye Tracker official Discord in one of their channels there. Since then I've been telling anybody who will listen about this amazing software!

Thank you for making this public and for developing such a great accessibility tool! It truly is invaluable.

Merry Christmas and Happy Holidays and Happy New Year!