PrecisionGazeMouse/PrecisionGazeMouse #50: Is "continuous mouse movement" supported?

rubenchristoffer, 2020-10-17:
I was wondering if "continuous mouse movement" is supported, meaning that if I move my head to the side (or up and down) it moves the cursor until I move my head back to the original resting position again. This would be great for games for instance.

rubenchristoffer, 2020-10-17:
What would also be cool is if the cursor moves faster the further you turn your head away from the resting position, kind of like a joystick would do

nash8350, 2020-10-18:
@rubenchristoffer yes it does offer a "continuous mouse movement mode" you can find more detail on it here https://precisiongazemouse.org/manual/

Your idea to move the cursor faster sounds similar to this issue https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/issues/9. I am looking for volunteers if you're interested.

rubenchristoffer, 2020-10-19:
Reading my question again, perhaps i wasn't clear enough. From what I gathered from [this](https://www.youtube.com/watch?v=htbBH0W5F78) video, the mouse warps to where the head is pointing (absolute mouse movement). I was wondering if it is possible to use relative mouse movement, so that if my head is pointing to the side, the cursor _will not stop moving_ until I return my head to a resting position. Is there such a feature?

nash8350, 2020-10-22:
Oh interesting I might call that a "joystick mode" or something like that. You're the first person to request it. It supports a few different head trackers that behave differently. eViacam and SmartNAV don't support that AFAIK, but you can check with them. EyeX head tracking is not very good. TrackIR would work because we get the API input directly, so it should be possible to code up.

rubenchristoffer, 2020-10-22:
I see. I just got a TrackIR 5 today, so any solution involving that would be cool. By the way, "joystick mode" is a great name!

nash8350, 2020-10-26:
Great I added a help wanted label to indicate that this could be a good feature for someone to help with. I'd be open to adding it if someone writes the code.