PrecisionGazeMouse/PrecisionGazeMouse #51: TrackIR Only mode not working

rubenchristoffer, 2020-10-22:
So I got the TrackIR 5 today and set it up, but I couldn't get it to work with PrecisionGazeMouse. Whenever I move my head, the cursor does not move. Perhaps I did something wrong or missed something?

Here's what I tried:
- Running TrackIR in administrator mode (same as PrecisionGazeMouse)
- Used both continous mode and the other mode
- Tried different sensitivity values (didn't seem to make a difference when I adjusted it)
- I used the latest version of TrackIR (5.4.2)
- I used the latest version of PrecisionGazeMouse

Here's what I noticed:
- I couldn't find the profile "GazePlusMouse" in the titles tab of the TrackIR 5 software, so I wasn't able to change it to One:One profile as stated in documentation
- It says "running" as status in PrecisionGazeMouse and occasionally teleports the mouse somewhere
- The Head Rotation value shown in PrecisionGazeMouse does not change when moving my head

I don't have an eye tracker, so I didn't test any of the other modes. Any idea on how I could get it working?

nash8350, 2020-10-24:
You have to be able to select "GazePlusMouse" because that is associated with a profile ID that we load in the API. Did you check for game updates? I tried updating mine and I still see it in the list.

rubenchristoffer, 2020-10-24:
Thanks, got it working now after updating game list.

closed 2020-10-24 by rubenchristoffer

seanconnors1000, 2023-05-31:
I'm having the exact same issue, except I don't play games. I just need it because I have RSI. Where would we find the "GazePlusMouse"? I don't see it in any menu on the program