PrecisionGazeMouse/PrecisionGazeMouse #52: Cursor often freezes for 1-2 seconds

rubenchristoffer, 2020-10-24:
So I got the cursor movement to work, but I noticed that the cursor often freezes for 1-2 seconds even when I am moving my head and the movement is displayed inside the TrackIR 5 application. The mode I'm using is "TrackIR only". It also seems to only happen when either TrackIR 5 or PrecisionGazeMouse is minimized / not in focus. Any reason why this might be?

nash8350, 2020-10-25:
Tough question we need to narrow down possible causes. So the movement shows in the TrackIR app. Does it also update the gaze or head position in the text field on the PGM app? I think normally it updates every second or more. Are both show warp bar and show gaze tracker disabled? Is your CPU and IO usage low in the task manager so there is no resource contention?

rubenchristoffer, 2020-10-25:
I found out that PGM actually goes into "Pause" mode (status) for some reason. Again, it only seems to happen when either program is out of focus. Both warp bar and gaze tracker options are disabled. It shouldn't be a resource starvation problem as resources never max out at 100% and I tried setting TrackIR and PGM priority to realtime and the problem persists.

nash8350, 2020-10-26:
PGM pauses when the pause key is pressed or when when another device moves your mouse https://precisiongazemouse.org/manual/. Is it possible either of those things is happening?

rubenchristoffer, 2020-10-27:
I found that it was indeed a program causing problems, even though it wasn't moving the mouse. Thanks for the help.

closed 2020-10-27 by rubenchristoffer