PrecisionGazeMouse/PrecisionGazeMouse #53: Will this work with the Tobii 5?

Ziriax, 2020-11-23:
Because of shoulder and tennis + golf elbow rsi problems I want to give this a try. 

Will this work with the newest Tobii eye tracker 5?

I also see that Logitec has a really good 4K webcam, will this help too? Or is a standard 1920x1080 webcam good enough?

nash8350, 2020-12-04:
Yes I use it with Tobii Eye tracker 5. I have a 4k webcam but I think a standard webcam would work fine.

closed 2020-12-27 by nash8350

Ziriax, 2020-12-27:
Thanks for your feedback!

canyoubarrett, 2021-01-25:
thank you for making this. I am getting a tobii 5 to use this. I cant tell you how many peripherals i have tried. Thank you!

Ziriax, 2021-08-24:
@canyoubarrett Are you happy with the Tobii 5 and this software? I'm planning to buy one myself.