PrecisionGazeMouse/PrecisionGazeMouse #54: Support for Trackerpro 2 and Headmouse Nano

ontwikkelfabriek, 2020-11-28:
Does the head tracking part support the:
- trackerpro 2 
- headmouse nano

https://www.ablenetinc.com/trackerpro-2
https://www.orin.com/access/headmouse/

nash8350, 2020-12-04:
I don't personally own either of these devices so I can't test myself, but they look similar to SmartNav. You could try running it in SmartNav mode.

ontwikkelfabriek, 2020-12-07:
I have a Smartnav too and works great. But considering they cancelled production I was looking for alternatives in case mine dies. And from what I read, the trackerpro and headmouse nano are on par or better with pure head tracking then the Smartnav.

nash8350, 2020-12-27:
Aww that's a bummer they stopped making SmartNav. The TrackIR is a similar IR mouse that seems to work well. 

When I suggested running in SmartNav mode, I mean these two products look like they might be compatible with Precision Gaze Mouse if you select that mode in the dropdown box of Precision Gaze Mouse. This mode assumes precision mouse control from an external program, which I believe these products provide. If it works, let me know!

nash8350, 2021-01-30:
I didn't see a response, so I'm closing this issue. Please reopen if you have anything to share, like if it worked for you.

closed 2021-01-30 by nash8350