PrecisionGazeMouse/PrecisionGazeMouse #55: Doesnt recognize input from 4c eyetracker

aaaak4, 2021-01-06:
Hi I tried installing the software but it doesnt recognize the input from my Tobii Eyetracker 4C but could only find a disabled webcam in my Lenovo laptop. what do I do? (tried to enble the webcam through privacy settings which just crashed the laptop with "PFN List Corrupt"

aaaak4, 2021-01-06:
I made it more precise by changing it to Eyex and Smartnav even though I dont have a Smartnav. Now im mostly wondering how you moved the mouse within the circle or how to make the circle smaller

nash8350, 2021-01-30:
> the cursor is not at all at the same place as the gazetracker

Can you provide more detail on what you mean? Is it in random places, or always offset by some amount? If you click "Show gaze tracker" you can see circles to showing the internal state of the system. Also, what modes do you have enabled?

> it works with standard tobii if I click for movement 

What is standard tobii? Are you talking about the feature in the tobii experience app "warp mouse pointer to where you look"?

phbuckbee, 2021-02-05:
Firstly, thank you for making this! I am very excited to use this once I get the kinks sorted out. I am having a similar issue. I am using tobii eye tracker 5 and the trackir. When I am using Precision Gaze Mouse, the cursor is always offset by the same amount wherever my eyes move. I attached a screen shot that shows my actual gaze on the tobii eye tracker 5 and my gaze on the Precision Gaze Mouse Gaze Tracker.
[image]

nash8350, 2021-02-14:
@phbuckbee I see in your screenshot that the status is "No TrackIR connected". This can happen when when Precision Gaze Mouse cannot access the TrackIR API. Did you follow the TrackIR instructions here https://precisiongazemouse.org/manual/? If it cannot access TrackIR, it will just leave the mouse at origin, which is the upper left corner of the screen. You can also check whether its working using one of the other Tracker Modes.

phbuckbee, 2021-02-14:
I disconnected it because my computer was getting a little laggy with a few other programs I was running. It does the exact same thing when the TrackIR is connected. When I initially set up the TrackIR, I used the instructions provided in the link in your comment. I'd be happy to try and set up a zoom meeting or something along those lines so you are able to see it first hand if you would like, or, if I can provide feedback in another way, let me know. In the meantime, I'll try uninstalling and reinstalling the TrackIR software. Would you recommend using the eViacam w/ the Tobii Eyetracker V in the meantime?

nash8350, 2021-02-20:
Does the TrackIR say the title is GazePlusMouse and PGM says "Running"?

[image]

You can use eViacam with PGM. Just select the eViacam mode in the dropdown. Its better than with standard tobii because it handles mouse clicks for you, and it can pause eViacam when you are not pressing the hotkey.

phbuckbee, 2021-02-23:
Yes the TrackIR says that title is GazePlusMouse and PGM says "Running".

phbuckbee, 2021-02-24:
It seems the problem might be geared more towards the Tobii Eyetracker 5. When using the eyex only option on precision gaze mouse, the result is extremely similar to the screenshot posted above. What software / application is required to run the tobii eyetracker 5 with the precision gaze mouse program? I just want to make sure I actually have it set up in the correct manner.

nash8350, 2021-02-26:
Just the regular tobii drivers. I have a 5 and it works fine for me. I wonder if there is something different about how the gaze point maps to your screen bounds? Could be something with how your display is set up. The Gaze Position in PGM should say (0,0) when you look to the top left, and match your screen resolution when you look to the bottom right.

phbuckbee, 2021-02-27:
Okay. I'm definitely have negative numbers show up on the x-axis values for the gaze position when running the eyex and trackir.

phbuckbee, 2021-02-27:
The mouse/gaze tracker never seems to want to stay in the top left corner too. After a few seconds to mouse and gaze tracker both drop 2-3 inches.

phbuckbee, 2021-03-02:
If it helps, I use a laptop with a separate monitor connected via HDMI so I can use two screens (aka one continuous screen). When using precision gaze mouse, I change my setup so that only one screen is being used. The external monitor is what I have been trying to use precision gaze mouse with so I do not have to significantly elevate my laptop to meet my natural eye level.

nash8350, 2021-03-12:
Interesting that might be it! I have a separate monitor and it works for me if I select "Show only on display 2" but if choose "Extend this display" then it doesn't account for second screen properly. Can you try changing to "Show only on display X" where X has Tobii attached to see if it works better? You might have to restart PGM and TrackIR because I'm not sure if they account for the new screen bounds dynamically.

Also, is having support for extended screens important for you? That could be a new feature. I haven't added it yet because Tobii only supports one screen at a time, but it seems to handle the extended desktop use case more gracefully than PGM.

phbuckbee, 2021-03-12:
I have some positive news: I managed to get PGM to start working on my laptop. I didn't knowingly make any changes so I'm not sure why it wasn't working initially.

In regards to you message, I did the same thing, and I will try it one more time to see if it will work on the second screen. Maybe it will start working like my laptop did! Oh, not restarting PGM and TrackIR might've been my issue to begin with now that you mention it.

Having two screens helps me a lot at work and with school related tasks. Being able to use PGM with two monitors would be even more helpful. A couple of my co-workers are interested in using PGM, and they were hoping to be able to use it with two monitors.

nash8350, 2021-04-10:
That's great I'm glad its working for you now! Maybe it was just a matter of restarting the apps.

Not sure if you're aware but EyeX has a limitation that it only works on a single monitor https://help.tobii.com/hc/en-us/articles/209529429-Can-I-use-multiple-screens-

When you say you're hoping to use PGM with two monitors, would it still be useful to have PGM on the first monitor and then use your regular mouse & keyboard on the second monitor? Perhaps you could use the second monitor as a display you don't need to interact with very often? I only use a single monitor normally, so I'm not sure how common this use case is.

Eduard2798, 2021-04-16:
I am having a similar issue using Tobii Eye tracker 5 and PGM with EyeX only mode. When moving my gaze, the cursor is always offset towards the center, and tends to fall outside of the grey circle in the overlay. This does not happen if I switch to EyeX and SmartNav mode (I don't have a SmartNav, so then I cannot move the cursor with my head in this case). On top of that when moving the cursor with my head, the cursor moves outside the grey circle (with high sensitivity), not sure if that is intended.

I am using a single monitor and have tried restarting the computer and the apps.

nash8350, 2021-05-16:
Closing because it seems like its working now. I see there is a request for two monitors but this is not supported by Tobii. Feel free to open another issue if needed.

closed 2021-05-16 by nash8350