PrecisionGazeMouse/PrecisionGazeMouse #56: Any chance for a machine installer?

teknowledgist, 2021-01-25:
Nice program, but the installer is a "click-once" (i.e. per-user) installer.  

I understand why that is attractive for a segment of users, but would it be possible to create a machine installer that could be installed once (by an admin/SYSTEM) for all users of the machine?  I would also really prefer a "silent" install option/switch to allow for centralized deployment.  Obviously, the software would need to be able to accommodate settings being held under each user's profile, so I guess this isn't just a matter of building the installer.

Thanks.

nash8350, 2021-01-30:
> would it be possible to create a machine installer that could be installed once (by an admin/SYSTEM) for all users of the machine? I would also really prefer a "silent" install option/switch to allow for centralized deployment.

Sure it sounds possible. I'm curious to hear about the use case driving this request. Normally these features are needed when you expect many users on a single machine, and across many machines. This is often used by people with disabilities and there are not many of us. How many users are you thinking of?

> the software would need to be able to accommodate settings being held under each user's profile

The settings are stored in each user's own folder already. I imagine this isn't a big issue, but let me know if I'm missing something.

teknowledgist, 2021-01-30:
I don't have a strong argument for it, really, but a few weak ones:

1. I work for a Library where we could potentially use PGM (but TBPH, we don't have plans at this time).  Since users log in individually, we would want a system install over a per-user install.
2. Most schools/libraries/organizations use centralized management/deployment of software, so it makes sense to offer something.
3. I manage a number of Chocolatey packages (https://chocolatey.org/profiles/teknowledgist) which make it easy to install software of all types.  I just finished a package for eViacam and noticed the reference to your software.  My wife is an OT, so I'm moderately aware of the needs of the differently-abled, and I thought PGM would make a nice package too.  However, Chocolatey packages are supposed to be machine installs, not user installs.
4. As a sysadmin for a sizeable number of workstations and servers, I personally dislike user-installs.  They make sysadmins who want (or need) to better control the user experience and restrictions pull out their hair.  

So, if you choose not to make a machine installer, I guess no real harm done, but if you do, maybe some more folks get to use (or maybe just be aware of) Precision Gaze Mouse.

I can't say of you are missing anything as far as settings.  I'm not sure how the settings are referenced.  For a machine install, only the installer writes to the binary package location and/or the HKLM registry.  All adjustments/settings/files that are saved/written by the application itself must be within the unprivileged user profile either as files or in HKCU.  Some programs will allow for both system settings and user settings with rules about how they resolve conflicts.  

Thanks for the follow up.

nash8350, 2021-02-14:
Cool, well I'd be open to a Chocolatey package if it will help more people use it. How would you get the Tobii EyeX device installed on the library workstations? It sounds like you don't have plans at this time, but if you do, let me know. I've never built a Chocolatey package, so it'd be great to have your help. I'll mark this issue as help wanted for now.

teknowledgist, 2021-02-15:
As I said, we don't have plans at the moment, nor do we have any Tobii eye trackers at this time.  We do try to accommodate different abilities when we can like chair accessible tables, screen readers and a braille printer.  I just built a package for eViacam and I'm just taking one step further.  😄 

As for a Chocolatey package, as the author of the software, you would have the right to have public repository admins give you maintainership, and I'd be willing to advise if you need it.  Otherwise, as long as you can build PGM to work for all users of a machine (i.e. user-profile based settings) and if you use an installer you include a silent install option, I can build/maintain the Chocolatey package.  The binary (installer or standalone) would need to be downloadable as only official binaries can be on the public repository (not that I have the skill to compile anything anyway).

nash8350, 2021-02-20:
Would an MSI installer work?

teknowledgist, 2021-02-20:
Possibly.  Again, I don't know how the internals of the software work, so it's difficult to say for sure.  

If the MSI installer has an `ALLUSERS` option that installs to `C:\Program Files` (or `C:\Program Files (x86)`) **and** PGM stores/references settings/logs in the user profile and doesn't attempt to write to it's own write-protected directory or the HKLM registry, then I think an MSI installer would work.

TheNameIsNigel, 2021-04-12:
> Would an MSI installer work?

MSI installers would work best in the use case as libraries and other institutions w/ large amounts of Windows PCs can easily deploy the MSI via Group Policy and it would just take less than 2 minutes of work initially to get it added to GPO and it'll auto-deploy to the rest of the PCs.

teknowledgist, 2021-04-13:
Yes, an MSI installer would make deployment very straightforward, but it would be pretty moot (for deployment) unless the software is compatible with a machine-level install on multi-user systems.

TheNameIsNigel, 2021-04-13:
> Yes, an MSI installer would make deployment very straightforward, but it would be pretty moot (for deployment) unless the software is compatible with a machine-level install on multi-user systems.

If you’re deploying an MSI, it’s always going to be in the root Program Files directory and most apps write to user’s AppData folder. I’m unsure of what you’re asking for because distributing a MSI via GPO is how you do it.

teknowledgist, 2021-04-13:
> If you’re deploying an MSI, it’s always going to be in the root Program Files directory and most apps write to user’s AppData folder. 

In a sysadmin's ideal world, those are correct, but in my experience, you can't assume either one of those.  I've been annoyed far too many times by software (typically, "small shop"/specialty software) that does not follow those conventions.  If building an MSI for PGM is all that is needed for a functional machine-install, great, but it's unclear to me if the current version looks/works in the current user's profile or just the installing user's profile.

> I’m unsure of what you’re asking for because distributing a MSI via GPO is how you do it.

Deploying applications via GPO was popular 10-15 years ago and still (kinda) works, but it is rarely recommended these days.

TheNameIsNigel, 2021-04-17:
Not sure what you mean by rarely recommended, most major companies still use it, either deploying via GPO or Software Center. Also, Active Directory is used in almost all major corporations (~90-95% of Fortune 500 uses it), I have yet to find a company that still uses eDirectory or other alternatives. Deploying a MSI via GPO or Software Center is the easiest way that requires little setup time and almost zero end-user steps needed for deployment.

TheNameIsNigel, 2021-04-17:
Most major companies still use it, either deploying via GPO or Software Center. Also, Active Directory is used in almost all major corporations (~90-95% of Fortune 500 uses it), I have yet to find a company that still uses eDirectory or other alternatives. Deploying a MSI via GPO or Software Center is the easiest way that requires little setup time and almost zero end-user steps needed for deployment.

That being said, I do agree with what you said.

euridium, 2021-09-10:
Hello,
has there been any additional development on off-line installer?I need to install this application on a machine that is Air gap and does not have access to the Internet, Is there a straightforward approach to use the application in this way as it currently stands?
I am using E via cam and voice dictation at the moment, and I would like to try using A Tobii 5, But I am reluctant to buy one if I will not be able to get it to work
I do have a demo machine that is connected to the Internet, so I could download it onto that one.

Any ideas?
thanks

Matt-tf, 2021-09-18:
@euridium: I was also curious if it was possible to install on an offline machine. Did you check or find if there is a way to install a ClickOnce application offline? According to this article it should be possible if you have the "Application Files" (and the .application file) [https://stackoverflow.com/questions/5618209/install-a-clickonce-application-on-an-offline-machine](url). But I think the files would have to be provided.

nash8350, 2021-10-10:
You can also just copy the executable files to a directory, they should work fine with no installer.