PrecisionGazeMouse/PrecisionGazeMouse #58: can this work with Tobii 5

IVNtech, 2021-02-09:
Mr Henry, first of all, WOWW , your work is amazing I want to congratulate you! ,   I was wondering if the code can work with TOBII 5 , I read the WHY and I am in total agreement with you I got the same problem with the hand , after hours of writing I got my hand with pain, special the right hand because its the mouse hand,  moving the mouse from keyboard to mouse is so stressful and feel lazy and stupid moving my hand from one place to another ( I DONT WANT TO LEAVE THE KEYS!! ), for sure I want to keep looking on this repository, please keep working on it!! I know this can be a BOOM

I want to know if this can work with tobbi 5 because is IMPOSIBLE TO BUY any other toby product , because its not in sale the products just the tobbi 5 And the specifications tolds that its for gaming, and I never play a game on the computer just love to CODE.

Regards, Daniel from IVN

nash8350, 2021-02-14:
Yep it works great on Tobii 5. That's actually what I use myself. I hope it hepls you. I'll close this issue since I think I answered your question, but feel free to open another one if you have a problem.

closed 2021-02-14 by nash8350

IVNtech, 2021-02-15:
Mr.Henry @nash8350 , thanks for your fast response, I have a last question, do I need to buy the TrackerIR for a better result ? or the TOBII 5 makes now all the work? thanks a lot for your time

nash8350, 2021-02-20:
@IVNtech I like using it with eViacam, which is a free software and doesn't require you to buy anything but a webcam