PrecisionGazeMouse/PrecisionGazeMouse #59: TrackIr 5 Not Working

YaTvoyKapitan, 2021-03-09:
My TrackIr 5 is shown to be operational using its 5.4 edition software, and the Precision Gaze Mouse recognizes it, but the head rotation is stuck on (0, 0). I guess I'll try using my WebCam for now.

nash8350, 2021-03-12:
Thanks for reporting this! I see the same problem on my end as well. Its possible they updated the API since I originally wrote this, and I might need to update the code. I'll mark this as a bug. 

To be honest though I never got good results with their head tracking and I find the webcam much more accurate. Was it working well for you before? Maybe it will work better with this new update.

nash8350, 2021-03-12:
Sorry I misread your comment. I thought you were talking about EyeX head tracking and I discovered this bug https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/issues/60

I see now that you are using TrackIR head tracking. I just tried on my PC using version 5.4 and its working for me. Did you read the user manual for TrackIR? https://precisiongazemouse.org/manual/ What does the status field say in PGM? If its successfully connected it will say "Running"

[image]

nash8350, 2021-04-10:
Its been about a month and I didn't get a response so I'll close this. Feel free to reopen if you want to pick up again.

closed 2021-04-10 by nash8350