PrecisionGazeMouse/PrecisionGazeMouse #60: EyeX head tracking not working

nash8350, 2021-03-12:
Choosing the "EyeX only" mode with Tobii head tracking is not working for me. Tobii may have updated the API since I originally wrote it. Marking this a as bug to fix.

The workaround is to choose one of the other modes for head tracking. eViacam currently gives the best results for me.

IVNtech, 2021-03-15:
For me too, I cant use the EyeX only, The best result for me is eyeX and eViacom, very cool and I am working on a 40 inch screen, how cool is that!!

Eduard2798, 2021-04-16:
I am having a similar issue using Tobii Eye tracker 5 and PGM with EyeX only mode. When moving my gaze, the cursor is always offset towards the center, and tends to fall outside of the grey circle in the overlay. This does not happen if I switch to EyeX and SmartNav mode (I don't have a SmartNav, so then I cannot move the cursor with my head in this case). On top of that when moving the cursor with my head, the cursor moves outside the grey circle (with high sensitivity), not sure if that is intended. 

I am using a single monitor and have tried restarting the computer and the apps.

nash8350, 2021-04-18:
I tried this again today and it appears to be working. Perhaps there was an update in the meantime.

closed 2021-04-18 by nash8350