PrecisionGazeMouse/PrecisionGazeMouse #61: smaller show gaze tracker

IVNtech, 2021-03-15:
Hi Nash, I get the Tobii 5 and install the precision gaze mouse and I got to say its wonderful software makes the JOB done! pretty well, so well I was wondering its there a way to put an option with a slider on the gaze tracker to make the GAZE smaller, I mean the area of the gaze tracker is too big and when I get out of that area the mouse pointer jumps out BUT its really to bigger, I think that if the gaze where smaller I cut to move the cursor faster than trying to move the head outside the gaze to relocate the mouse nearby, hope to make sense, Does there is any SDK for windows I would love to try to tweak some options. 

The rest is so cool Mr.Nash thanks a lot for making this, and by the way, I have the same problem as you the one that moving the hand from the keyboard to the mouse is a waste of time... and a lot of pain on the hand

nash8350, 2021-04-10:
> an option with a slider on the gaze tracker to make the GAZE smaller

I want to make sure I understand what you mean by GAZE. When you turn on the "Show gaze tracker" option it should show a circle on the screen. If you look outside that circle, the mouse warps to the new location. You want to make that circle smaller?

There is no way to change this in the UI yet, but I have a workaround. You can change the setting by editing your user.config file. For me, its stored at `C:\Users\Admin\AppData\Local\PrecisionGazeMouse\PrecisionGazeMouse.exe_Url_ilksjdflkdjsfkldjfdf\1.0.0.0`. You can add these lines to set the threshold. The default is 200, but you can make it smaller.
```
            <setting name="EyeXWarpThreshold" serializeAs="String">
                <value>200</value>
            </setting>
```

Let me know if this is what you mean, and if this helps. If so, I could consider making it a full UI setting.

douglaz, 2021-04-13:
Started using today. This setting seems to be useful. TOBII 5 is very precise, so we need to trust it more instead of eviacom

nash8350, 2021-04-18:
Great to hear! I'll mark this as help wanted. I might be able to add it next time I get into coding.

IVNtech, 2021-04-18:
please MrNash add the slider for the gaze smaller or to set up the size each user want it, I try the code and is awesome to have it a little bit smaller because am on a 40-inch monitor so the gaze circle is very big BUT on my laptop, it works great so depending on each screen it changes the setup, the rest its amazing to work with the tobbi 5 and you program I could achieve the goal to NOT use anymore my mouse, thanks a lot!

nash8350, 2021-05-16:
I just added a track bar to the UI to adjust the sensitivity. Check it out in v1.16. Let me know if you have any problem with it.

IVNtech, 2021-05-16:
thats sound wonderfull thanks a lot thanks!

nash8350, 2021-06-11:
This is fixed so I'm closing this issue.

closed 2021-06-11 by nash8350