PrecisionGazeMouse/PrecisionGazeMouse #66: Click doesn't work for all programs

douglaz, 2021-04-15:
For instance, it doesn't work on virtualbox and on the vertical scroll bar of the slack app.

nash8350, 2021-04-17:
Interesting I am able to click the menu within VirtualBox 6. Is there a specific feature that is not working? 

I am able to reproduce the scrollbar issue in Slack. When I click on the portion of the scrollbar that does not contain bar itself, nothing happens. When I click it with my physical mouse then it scrolls. However, I can click and drag scrollbar itself by double tapping my mouse button. I'm not sure why this happens since the click is a  low level event in Windows.

douglaz, 2021-05-03:
I found out the issue on virtualbox happens only if I map a special key, like LaunchApplication.

If I map F3, for instance, it works.