PrecisionGazeMouse/PrecisionGazeMouse #67: Mouse warp and head tracking issue with Tobii Eye tracker 5 and EyeX only mode.

Eduard2798, 2021-04-16:
I am having an issue using Tobii Eye tracker 5 and PGM with EyeX only mode. When moving my gaze, the cursor is always offset towards the center, and tends to fall outside of the grey circle in the overlay. This does not happen if I switch to EyeX and SmartNav mode (I don't have a SmartNav, so then I cannot move the cursor with my head in this case). 

Interestingly, if I choose a very low sensitivity setting, the cursor tends to fall much closer to the center of the grey circle after moving my gaze, but then the head movements barely move the mouse and are not usable. 

On top of that when moving the cursor with my head, the cursor moves outside the grey circle (with high sensitivity), not sure if that is intended.

I am using a single monitor and have tried restarting the computer and reinstalling PGM. 
[image]

nash8350, 2021-04-18:
It looks like EyeX is reporting that your head is rotated when it is not. I'm not sure if you are a programmer but you can check the raw data coming from EyeX here https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/master/PrecisionGazeMouse/PrecisionPointers/EyeXPrecisionPointer.cs#L122.

Personally, I cannot use EyeX because the accuracy is not very good. Have you tried eViacam mode yet? I find that one is more accurate.

nash8350, 2021-05-16:
Closing due to no response. Feel free to reopen if you have more to add.

closed 2021-05-16 by nash8350