PrecisionGazeMouse/PrecisionGazeMouse #69: Hide cursor for compatibility with gaze communication software

AngelAspace, 2021-06-01:
Thank you very much for rhis software, it might be very useful for cerebral palsy people. 
Sometimes they use software communication with big squares, then high precision is not required
In this cases they select standing on the item for the time required. The whole square is highlited so cursor is not necessary and in some cases catches the gaze of the user
This is why, if it is possible, it would fine to have a check for visible mouse.

nash8350, 2021-06-11:
If you don't need precision why not just use Tobii's built in mouse feature?

AngelAspace, 2021-06-11:
> Enviat el: divendres, 11 de juny de 2021 15:35
> Per a: PrecisionGazeMouse/PrecisionGazeMouse ***@***.***>
> A/c: Àngel Aguilar Morales ***@***.***>; Author ***@***.***>

Hello thank you for your response, do you mean Gaze point?
The reason is that gaze point is no longer available for the new game tobiis. (since april)
Regards

ÀNGEL AGUILAR MORALES
Director del Servei de Recursos i Tecnologia de Suport  d'Aspace
Telf. +34 933258300
***@***.***  /  www.aspace.cat

[cid:5bc59582-1e3d-4a4b-ba4b-2c66e4af5cf5]

nash8350, 2021-08-10:
Have you considered using a program like AutoHideMouseCursor http://www.softwareok.com/?seite=Microsoft/AutoHideMouseCursor?