PrecisionGazeMouse/PrecisionGazeMouse #76: Option to disable mouse pointer movement when mouse button is down

Ziriax, 2021-09-17:
First of all thanks for this nice software.

I just purchases a Tobii 5, to see if it can reduce the RSI I got after programming for over 40 years now.

So I'm using this software in combination with a mouse, to reduce the amount of motion I have to do with it.

One thing I noticed that when I'm dragging a slider (so I have the mouse button pressed), to pointer is still following my eyes.

However, most of the time this doesn't seem what you want: when dragging a slider, you typically look elsewhere on the screen.

Do you think it would be an option to disable pointer movement whenever a mouse button is pressed?

Ziriax, 2021-09-17:
Wow I must say, this software really works well :)

I actually noticed that my suggested option should not just be for when the mouse button is down, but when the user is doing **any** mouse movement. User mouse movement should always override the tracking IMHO.

So for example, when I'm painting a large circle with paint with my mouse, the cursor should not jump to my eye position like it does now, because that makes it impossible to draw a circle 😉 Same when selecting text, etc...

But this software has so much potential IMO! Nice work.

nash8350, 2021-10-10:
@Ziriax thanks for trying it out! I had a similar issue of RSI. 

>when dragging a slider, you typically look elsewhere on the screen. 

In this case, why do you want to look somewhere else on the screen? Don't you need to look at the position of the slider in order to aim it and release at the right location?  A specific example might help me understand.

> Do you think it would be an option to disable pointer movement whenever a mouse button is pressed?

How would you drag the mouse if pointer movement was disabled?

> User mouse movement should always override the tracking IMHO.

It already does. When I move my mouse with my hand, it follows where my hand goes. It only goes back to eye tracking when I stop moving my mouse with my hand.

>  when I'm painting a large circle with paint with my mouse, the cursor should not jump to my eye position like it does now, because that makes it impossible to draw a circle 

Not sure I understand your example. Are you painting a circle using precision gaze mouse or the mouse in your hand? If you are using precision gaze mouse the cursor should move with your eye gaze so you can fill in the circle. If its a small circle you can use the precision pointer control within an adjustable radius. For large circles, you'll want to warp the cursor beyond the range of the precision pointer.

nash8350, 2021-10-30:
Closing due to no response. Please reopen if you'd like to continue discussing it.

closed 2021-10-30 by nash8350

Ziriax, 2021-10-30:
Sorry, I forgot to reply. I'm using my own software now, based on the C# example from Tobii.  Since I don't have a license to use calibrated input (or for some reason I get imprecise data from the Tobii stream engine), I used ML.NET to train the mapping from gaze points to mouse points. Improves a lot. But gaze control still isn't pixel perfect, but close to 10 pixels, depending the area on the screen. Thanks for your software, it really inspired me.

nash8350, 2021-11-29:
@Ziriax Thanks for letting me know! I get imprecise data from Tobii too, and its only accurate to about 10 pixels as you mentioned. I combined it with head tracking to get down to 1 pixel accuracy, hence the name "precision gaze mouse".