PrecisionGazeMouse/PrecisionGazeMouse #78: Please help a quadriplegic

TomtheQuad, 2021-10-18:
I’m a quadriplegic who can’t use my hands at all.

I used SmartNav by naturalpoint for about 20 years. It was great. But they discontinued the product a couple of years ago and my last unit finally died on me about a month back. I have been looking for a good alternative with no good solutions presenting themselves so far. I tried an eye tracker from Tobii/dynavox and it didn’t really work well. I was intrigued by the video of the Precision Gaze Mouse, particularly since it could be used with NaturalPoint’s TrackIR. I ordered a TrackIR and it arrived today.

The user manual for Precision Gaze Mouse says that I have to choose “GazePlusMouse” in the “titles tab” of the TrackIR but I find no entry for anything like that, nor anything that could be confused with the Precision Gaze Mouse in TrackIR. Nor was there an option for me to choose a new title to enter manually.

Please advise. I’m at my wit’s end. I had great computer facility for about 20 years with the SmartNav and right now I feel dead in the water. I’ve tried multiple options already and nothing else gets me more than about 1/3 as effective as I was with the SmartNav.

Please help. I want my computer capability back :-(

nash8350, 2021-10-19:
Sorry to hear you lost your computer capability! I'd be really frustrated if that happened. Did you try refreshing the titles list in the TrackIR program?

TomtheQuad, 2021-10-21:
> Sorry to hear you lost your computer capability! I'd be really frustrated if that happened. Did you try refreshing the titles list in the TrackIR program?

How do I do that?

Basically, to be clear, this is what I’m attempting to do: use a TrackIR ONLY (no eye-tracking) with the Precision Gaze Mouse software (and a 3rd-party dwell-clicking app) to control my mouse in a Windows desktop environment (browsing, clicking around, etc.). This is what I used to do with the SmartNav from naturalpoint (a product that has now, VERY unfortunately, been discontinued).

The demo video regarding the Precision Gaze Mouse looked closest to replicating the performance of the SmartNav, compared to everything else I’ve seen out there. And I have now tried some of the alternatives, all of which are very poor substitutes and extremely unappealing (so far. I have one more coming in tomorrow).

Please tell me EXACTLY what I have to do to make this work. Do I open Precision Gaze Mouse first? Does the order in which I open the applications matter?

How do I refresh the software list in TrackIR? All I see there are games (none of which I own). Is there a way to manually add an entry there?

I have the Trackclip Pro as well but I don’t care for it much. The TrackIR seems to pick things up fine if I put a large reflective dot on my nose (which I’m used to from the SmartNav), so I would prefer to keep doing that.

Please advise. This is one of my last 2 chances of regaining something close to normal mouse control. My quality-of-life will be undeniably worse if I can’t get back to something similar to what I enjoyed for about 20 years with the SmartNav.

Thank you in advance.

nash8350, 2021-10-30:
Great let me know how it goes

nash8350, 2021-12-16:
Closing due to no response, hoping you got it working

closed 2021-12-16 by nash8350