PrecisionGazeMouse/PrecisionGazeMouse #79: Does this software work with the Tobii Eye Tracker 5?

SpechtacularDave, 2021-12-07:
Does this software work with the Tobii Eye Tracker 5? I just bought one and installed the Tobii Experience program to calibrate/test out the tracker, and the device is working. But when I start precison gaze mouse I get "Error initializing Tobii EyeX..." and am unable to open the program.

I just bought the Tobii Eye Tracker 5 specifically to use with this program because the precision gaze manual listed "Tobii EyeX 4C or 5'", and I assumed the 5 referred to the Tobii Eye Tracker 5 because I couldn't find a current product called the EyeX 5. Is this correct? Or is/was the Tobii EyeX a different product? 

I've tried googling for a program called EyeX from Tobi, but haven't found anything yet.

Thanks for any help.

nash8350, 2021-12-16:
Yep it works with Tobii Eye Tracker 5. I should probably update the manual. They used to call their product EyeX a long time ago. It's probably worth making sure the eye tracker works with the Tobii software, then send a full error message.

leecannon, 2021-12-20:
I have the same issue, the Eye Tracker 5 works fine with Tobii Experience but PrecisionGazeMouse gives the error `Error initializing Tobii EyeX. Please install it before using Precision Gaze Mouse.` on open.

nash8350, 2021-12-27:
Huh interesting I use version 5 and it works fine for me. The line that produces that error is just the initialization for the Tobii EyeX Client library https://github.com/PrecisionGazeMouse/PrecisionGazeMouse/blob/19944070c279e424e0300fc99807f7c0b8a81387/PrecisionGazeMouse/Program.cs#L90. I don't log a stack trace there, but I could add a patch to print it out if you're willing to run it and report the output.

ShawnFumo, 2022-01-18:
Anyone have any luck with this? Henry, what version of Windows are you on? Maybe it isn't working on Win11?

And maybe this isn't the place to ask, but you said you're using version 5. Back in the example YouTube video, you mentioned head tracking was better with a separate device back then. Is the head tracking good enough with 5 to not need any other devices now?

Zakkai, 2022-01-26:
I have the same problem as leecannon with my Tobii 5. The weird thing is that it used to work fine until mid-December of 2021. I'm guessing this is because Tobii change their driver and it's no longer compatible with PGM. I'm using Windows 11.

Mattie, 2022-02-28:
Was considering installing this for T5 but seeing these comments, I'll wait until the compat is confirmed (Win11/Tobii5)

hernant, 2022-03-04:
Can confirm this works to me with Tobii 5 on Windows 10.

I remember something about having to install some beta drivers, but can't remember if this was needed to make it work with PresicionGazeMouse or just for Windows 10.

The following (in Korean, you'll have to translate it with your browser) has the links for said drivers: https://joytk.tistory.com/29

nash8350, 2022-03-11:
I use Tobii 5 on Windows 10 and it works fine for me. I have not tested it on Windows 11, but it sounds like there might be a driver compatibility issue. I don't have a copy of Win11 myself so I can't test it. I've heard about many issues with Win11 in general so I'm reluctant to upgrade right away. 

If folks can help narrow down the root cause of the problem that'd be helpful. @hernant has a good idea to try other programs that work with Tobii to see if its something that changed in the drivers Tobii uses or if its specific to PGM.

DidaMer, 2022-05-09:
Hi there, did anyone found any working fix for that ? I have the same error on windows 11 !

hernant, 2022-05-09:
So, I have Tobii 5 working on a PC with Windows 10 with all the software mentioned in the list above for about 2 years now.  I noticed a few weeks ago that a new software (called Tobii Experience) was installed replacing the one I had before to calibrate the Tobii device. Looks like this comes from the new driver that was released the past March 21. Everything continues working well with the exception that now Tobii doesn't appear as an Eye Tracking device in the Windows Settings. I understand this was possible to me only after I installed the beta driver 5.0.0.6035 and was finally cancelled and never released as a  final product. The new driver adds some integration with Windows though.

I tried to use the same Tobii device on a different laptop where it has been never used running also Windows 10. I installed the new driver directly and I get the same error described above on PGM and similar errors in the other software from the list. The device works well with the things it makes possible on Windows on its own, just not the software from the list.

I tried removing the new driver and installed the previous version plus the beta one I had before in my first computer. With this,  I can use/configure the device from the Windows's "Eye Control" settings. However I still get the same error with the other software from the list, plus I'm left without a manner to calibrate the device (maybe this was part of a different installer I never bothered to save).

So, I guess is possible that the problem occurs in brand new installations, not just Windows 11.

I'm willing to test things and see what is different in both computers. @nash8350, could you please add a patch I could use to generate the report you mentioned above?

nash8350, 2022-05-16:
@hernant I added better logging in release 1.17. Let me know if you can reproduce the issue. I'm also using the Tobii Experience and it continues working for me. I am not using a beta driver, but given that its still a beta there may be a problem with it. I assume the GA driver continues working fine in Windows 10?

sponer, 2022-06-11:
Anyone can confirm that's it working now?
I don't have the Tobi 5 and TrackIR yet, but planning to get them soon to use PrecisionGazeMouse
Thanks in advance

hernan-torrez, 2022-06-13:
I'm sorry for the very slow response.

I downloaded the version 1.17 and got the following alert message when ran from the laptop with the issue (tried with the offline installer and building from the source code myself):

[image]

"There was a problem starting the DLL 'Tobii.EyeX.Client.dll': The specified module could not be found".

I verified such DLL is in C:\Program Files\Tobii\Tobii EyeX\Tobii.EyeX.Client.dll

I compared the whole "Tobii EyeX" folder with the one in the computer where everything works well using diff tool. All files are equal, with the exception that the laptop with the issue has the following additional folder (and files):

[image]

To clarify, the laptop with the issue:
- Never had a Tobbi device before.
- Runs Windows 10.
- Has installed the last driver that came out in March 2022 (Tobii.EyeTracker5.Offline.Installer_4.143.0.19463).
- I get the error when I try the Tobbi device I have been using in other computer for about 2 years, I get a similar error in all other third party software that uses Tobbi (the out of the box features Tobii has work well). 

The computer where things work well had originally the driver: 4.124.0.15937 and a beta one (that adds more integration with Windows: Tobii.EyeTracker5.Offline.Installer_5.0.0.6035.exe. This year, around March or April the version 4.143.0.19463 got installed (either by a Windows update or the software itself).

frosk1, 2022-08-09:
I have bought the tobii eye tracker 5 and tested it on windows 11 and windwos 10.
Both versions did not work with Gaze Mouse and the current driver  Tobii.EyeTracker5.Offline.Installer_4.153.0.22216.

No I am lost and sitting on hardware, that I can not use unfortunately.

Gazemouse 1.17 is not starting, due to error message.

Will there be a patch coming soon to make Gazemouse work again @nash8350 ?

Because in the current state this software is not usable for any user, which makes this a serious issue!

hernant, 2022-08-09:
I'm sorry for the very slow response.

I downloaded the version 1.17 and got the following alert message when ran from the laptop with the issue (tried with the offline installer and building from the source code myself):

[image]

Text in Spanish says: "There was a problem starting the DLL 'Tobii.EyeX.Client.dll': The specified module could not be found".

I verified such DLL is indeed in C:\Program Files\Tobii\Tobii EyeX\Tobii.EyeX.Client.dll

I compared the whole "Tobii EyeX" folder with the one in the computer where everything works well using a diff tool. All files are equal, with the exception that the laptop with the issue has the following additional folder (and files):

[image]

Just in case, I also granted permissions to the folder to "Everyone".

To clarify, the laptop with the issue:

Never had a Tobbi device before.
Runs Windows 10.
Has installed the last driver that came out in March 2022 (Tobii.EyeTracker5.Offline.Installer_4.143.0.19463).
I get the error when I try the Tobbi device I have been using in other computer for about 2 years, I get a similar error in all other third party software that uses Tobbi (the out of the box features Tobii has work well).
The computer where things work well:

Runs Windows 10.
Had originally the driver: 4.124.0.15937 and a beta one (that adds more integration with Windows: Tobii.EyeTracker5.Offline.Installer_5.0.0.6035.exe. This year, around March or April the version 4.143.0.19463 got installed (either by a Windows update or the software itself).
All third party software, plus Tobbi features continue working well.

lavajci, 2022-10-05:
Was this ever confirmed to work on Windows 11 with the current Tobii software?

hernant, 2022-11-08:
Maybe related: [Optikey 4.0 will no longer include support for Tobii's gaming trackers (4C, ET5, etc) following an ongoing discussion with Tobii's legal department](https://github.com/OptiKey/OptiKey/issues/850#issuecomment-1216249404)

Loos kike the device you are suppose to use for non-gamming is Tobii Dynavox PCEye.